PiSupply/PiJuice #9: Python 2 / 3 Wrapper for software

shawaj, 2017-10-17:
We should create a Python 2 / 3 Wrapper for software package to make it easier to use within scripts in a python library

klint-k, 2017-10-19:
In this directory there is python script called pijuicetest.py: https://drive.google.com/drive/u/0/folders/0BzUkkXkOnknbOWsxb3ZpaGo1ZTA

It asks for login info. If I give my google login it is not accepted.

shawaj, 2017-10-19:
@klint-k I pasted the wrong link, fixed now

klint-k, 2017-10-21:
I have done this. Will Python3 work with the PiJuice, with just the print changed?
2to3 /usr/lib/python2.7/dist-packages/pijuice.py           
root: Genera (a bunch of the same looking stuff) txt
root: Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
RefactoringTool: Refactored /usr/lib/python2.7/dist-packages/pijuice.py
--- /usr/lib/python2.7/dist-packages/pijuice.py (original)
+++ /usr/lib/python2.7/dist-packages/pijuice.py (refactored)
@@ -138,7 +138,7 @@
                                #return {'error':'COMMUNICATION_ERROR'}
 
                if self._GetChecksum(d[0:-1]) != d[-1]:
-                       print self._GetChecksum(d), d, length
+                       print(self._GetChecksum(d), d, length)
                        return {'error':'DATA_CORRUPTED'}
                del d[-1]
                return {'data':d, 'error':'NO_ERROR'}
@@ -186,8 +186,8 @@
                                if (data==result['data']):
                                        return {'error':'NO_ERROR'}
                                else:
-                                       print 'wr', data
-                                       print 'rd', result['data']
+                                       print('wr', data)
+                                       print('rd', result['data'])
                                        # result['data']
                                        return {'error':'WRITE_FAILED'}
 
@@ -1056,7 +1056,7 @@
                        d[13] = (R >> 8)&0xFF
                except:
                        return {'error':'BAD_ARGUMENT'}
-               print d
+               print(d)
                return self.interface.WriteDataVerify(self.BATTERY_PROFILE_CMD, d, 0.2)
 
        batteryTempSenseOptions = ['NOT_USED', 'NTC', 'ON_BOARD', 'AUTO_DETECT']
RefactoringTool: Files that need to be modified:
RefactoringTool: /usr/lib/python2.7/dist-packages/pijuice.py

fix and copy to /usr/lib/python3.4/

tvoverbeek, 2017-10-22:
Requirement is same code has to run on both Python2 and Python3, since Python2 is still the default in Raspbian Stretch.
For print it is easy:
At the top of every source file include `from __future__ import print_function` and change all
print statements to the Python3 function type: print(....).
Other problems: when you want integer division you have to use '//'
Python3 all text is unicode and the str type is not the same as binary byte array.
See how the code in Papirus has changed.

klint-k, 2017-10-22:
The first problem that I did run into is without a PiJuice, I tried to run the test python as python3. 
I got the error;
print (pijuice.power.SetSystemPowerSwitch(0))
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
  File "/usr/lib/python3.4/pijuice.py", line 947, in SetSystemPowerSwitch
    return self.interface.WriteData(self.SYSTEM_POWER_SWITCH_CTRL_CMD, [d])
  File "/usr/lib/python3.4/pijuice.py", line 147, in WriteData
    fcs = self._GetChecksum(data)
  File "/usr/lib/python3.4/pijuice.py", line 63, in _GetChecksum
    fcs=fcs^x
TypeError: unsupported operand type(s) for ^: 'int' and 'float'
This error refers to PiJuiceInterface._GetChecksum()
In this instance what is the caret used for?
fcs = 0xff 
for x in data[::
    fcs=fcs^x
return fcs
I was thinking math I have not been able to google it properly. At first I thought that it might be a math type expression.

tvoverbeek, 2017-10-22:
> File "", line 5, in

The ^ operator is a bitwise operator (exclusive or).
Both fcs and x have to be integer.
Do not now why x in your case is a float.

klint-k, 2017-10-23:
@tvoverbeek is this a good fix?

`
        def _GetChecksum(self, data):
                fcs=0xFF
                for x in data[:]:
                        if type(x) == float:
                                x = int(x)
                                #print(x)
                        fcs=fcs^x
                return fcs`

tvoverbeek, 2017-10-23:
Would work. Suggest to use `if type(x) != int:`.
But it remains taking care of symptoms.
Need to investigate why the calling routines provide float in the data array which should provide integer values.

klint-k, 2017-10-24:
I tried to trace the data back the data. Got lost :( 
The print returned 0.0. 
I do have an other hat on the pi3. the i2c is at 68 though.
I wonder if it is kicking the 0.0?

tvoverbeek, 2017-10-25:
Went back to your original trace-back
```
Traceback (most recent call last):
File "", line 5, in 
File "/usr/lib/python3.4/pijuice.py", line 947, in SetSystemPowerSwitch
return self.interface.WriteData(self.SYSTEM_POWER_SWITCH_CTRL_CMD, [d])
File "/usr/lib/python3.4/pijuice.py", line 147, in WriteData
fcs = self._GetChecksum(data)
File "/usr/lib/python3.4/pijuice.py", line 63, in _GetChecksum
fcs=fcs^x
TypeError: unsupported operand type(s) for ^: 'int' and 'float'
```
Looking at SetSystemPowerSwitch:
```
	def SetSystemPowerSwitch(self, state):
		try:
			d = int(state)/100
		except:
			return {'error':'BAD_ARGUMENT'}
		return self.interface.WriteData(self.SYSTEM_POWER_SWITCH_CTRL_CMD, [d])
```
The problem is the statement `d = int(state)/100`.
In Python 2 this is integer division (with truncation to integer).
In Python 3 this results in a floating point number.
To force integer division in Python3 use the '//' operator.
To make the code with the '//' operator work in both Python 2 and 3 use the following line at the start of the module:
`from __future__ import print_function division`

shawaj, 2017-10-25:
I'm just going to reopen this because I still would ideally like this to be done at some point as part of PiJuice development. Even though for now there's not much to do.

tvoverbeek, 2017-10-25:
Waiting for my PiJuice to do the Python2/3 compatibility stuff.

klint-k, 2017-10-25:
@tvoverbeek Thanks. 
I went through on a editor that I was able to search for int( if it had a div. I change it to dbl div. I'll double cheek it when I get the board.

shawaj, 2017-10-25:
@tvoverbeek it'll be on its way in about a week I hope :-)

klint-k, 2017-12-02:
so is there a wrapper for python3 yet?

shawaj, 2017-12-03:
Not yet no. We will post here when there is

shawaj, 2017-12-11:
this is already built in python, so should not be a problem to sort this at some point soon.

klint-k, 2018-01-12:
Will this work for python 3.4 and up
```#!/usr/bin/env python
from __future__ import print_function 
from __future__ import division
from smbus import SMBus
import time
import datetime
import calendar
import sys, getopt
from ctypes import *
from fcntl import ioctl
import struct
import ctypes
import threading

pijuice_hard_functions = ['HARD_FUNC_POWER_ON', 'HARD_FUNC_POWER_OFF']
pijuice_sys_functions = ['SYS_FUNC_HALT', 'SYS_FUNC_HALT_POW_OFF', 'SYS_FUNC_SYS_OFF_HALT', 'SYS_FUNC_REBOOT']
pijuice_user_functions = ['USER_EVENT'] + ['USER_FUNC' + str(i+1) for i in range(0,15)]

class PiJuiceInterface():
	def __init__(self, bus=1, address=0x14):
		"""Create a new PiJuice instance.  Bus is an optional parameter that
		specifies the I2C bus number to use, for example 1 would use device
		/dev/i2c-1.  If bus is not specified then the open function should be
		called to open the bus.
		"""
		I2C_SLAVE             = 0x0703  # Use this slave address
		I2C_SLAVE_FORCE       = 0x0706  # Use this slave address, even if
                                # is already in use by a driver!
		I2C_PEC               = 0x0708  # != 0 to use PEC with SMBus

		self.i2cbus=SMBus(bus)
		self.addr=address
		#self._device = open('/dev/i2c-{0}'.format(bus), 'r+b', buffering=0)
		#ioctl(self._device.fileno(), I2C_SLAVE_FORCE, self.addr & 0x7F)
		self.t = None
		self.comError = False
		self.errTime = 0

	def __del__(self):
		"""Clean up any resources used by the PiJuice instance."""
		self.i2cbus = None
		#if self._device is not None:
			#self._device.close()
			#self._device = None

	def __enter__(self):
		"""Context manager enter function."""
		# Just return this object so it can be used in a with statement
		return self

	def __exit__(self, exc_type, exc_val, exc_tb):
		"""Context manager exit function, ensures resources are cleaned up."""
		#if self._device is not None:
			#self._device.close()
			#self._device = None
		self.i2cbus = None
		return False  # Don't suppress exceptions.

	def GetAddress(self):
		return self.addr

	def _GetChecksum(self, data):
		fcs=0xFF
		for x in data[:]:
			if type(x) == float:
				print(x)
				#x = int(x)
			fcs=fcs^x
		return fcs
[... 1201 lines cut ...]
		else:
			return {'data':{'version':format(ret['data'][0], 'x'), 'variant':format(ret['data'][1], 'x')}, 'error':'NO_ERROR'}

	def RunTestCalibration(self):
		self.interface.WriteData(248, [0x55, 0x26, 0xa0, 0x2b])
# Create an interface object for accessing PiJuice features via I2C bus.
class PiJuice(object):

	def __init__(self, bus=1, address=0x14):
		self.interface = PiJuiceInterface(bus, address)
		self.status = PiJuiceStatus(self.interface)
		self.config = PiJuiceConfig(self.interface)
		self.power = PiJuicePower(self.interface)
		self.rtcAlarm = PiJuiceRtcAlarm(self.interface)

	def __enter__(self):
		# Just return this object so it can be used in a with statement
		return self

	def __exit__(self, exc_type, exc_val, exc_tb):
		return False  # Don't suppress exceptions.

```

I have ran it without the PiJuice. There were no errors.

tvoverbeek, 2018-01-12:
After a quick inspection I would say it should work with python 3

shawaj, 2018-01-24:
Guess we can probably close this issue now?

klint-k, 2018-01-24:
Is it going to be installed and updstable?

klint-k, 2018-01-24:
Is it going to be installed and updatele?

shawaj, 2018-01-25:
@klint-k what do you mean?

tvoverbeek, 2018-01-25:
Current packages in the repository only run with python 2.7.
@klint-k did some changes to pijuice.py (see above) to make it both python 2 and 3 compatible.
Propose to let the dust settle on the current release (and the first Kickstarter shipment).
Then make all python software (pijuice.py, pijuice_sys.py, pijuice_gui.py, pijuice_tray.py) compatible with both python 2 and 3.
Should not be too much work, but let us first get the current version really ready.

shawaj, 2018-01-25:
Sounds like a plan

DenisKhrutsky, 2018-03-02:
Some changes were made to all scripts to support both versions of Python. @tvoverbeek, @klint-k, let me know if something is wrong

tvoverbeek, 2018-03-03:
Tried to run most of the python scripts manually under Python 3.5.3 (Stretch).

pijuice_sys.py has a problem on line 185 in Python 3:
`open(PID_FILE, 'w').write(pid).close()`
Replace with:
```
with open(PID_FILE, 'w') as f:
      f.write(pid)
```
I doubt if this worked in Python 2.
With this change it runs fine under Python 3.

pijuice_gui.py runs fine under Python 3
Implies pijuice.py is also fine.

Have not been able to run ppijuice_tray yet (problems interfacing with Gtk and Dbus).
pijuice_cli also runs fine under Python3 (with the change of Issue #69 )

Caveat: Only limited testing, could still be some bugs lurking somewhere.

We need separate packages for Python2 and Python3.
Will raise a separate issue for this.

DenisKhrutsky, 2018-03-05:
Fixed file handling (using with statements), issue #69 and tray icon methods (specifically set_tooltip -> set_tooltip_text and stub for set_blinking as I haven't found the replacement in Python 3 gi.repository).

shawaj, 2018-05-18:
Closing this as issue #70 addresses the same issue

closed 2018-05-18 by shawaj