PiSupply/PiJuice #10: Pogo???

klint-k, 2017-10-27:
I do like the info on the README. I am still wondering why there is a pogo in the package? Where is it meant to go?

shawaj, 2017-10-27:
This documentation is very very unfinished. It'll all be explained. It is
to interface with the run pin on the RasPi to enable auto restart when the
user has got power plugged into the RasPi micro USB port.

Thanks

tvoverbeek, 2017-10-27:
For more info on mounting the pogo pin see https://www.pi-supply.com/make/papirus-assembly-tips-and-gotchas/
The pogo pin is connected to the RTC alarm output on the HAT and to the reset input on the Pi.
This can be used for example to boot the Pi in the morning.
See also the RTC-Examples folder on github (https://github.com/PiSupply/PaPiRus/tree/master/RTC-Hat-Examples)

shawaj, 2017-10-27:
Thanks @tvoverbeek :-)

Only small difference to note is that the RTC on PiJuice is actually built into the ARM cortex m0 processor it's not using a standalone RTC chip like on PaPiRus. Functionality is practically identical though :-)

tvoverbeek, 2017-10-27:
Sorry, did not look good enough to notice this is abou the PiJuice ;-)

klint-k, 2017-10-27:
Cool I will see if it is TP1 or not.

shawaj, 2017-10-27:
What do you mean?

tvoverbeek, 2017-10-27:
Looking at the picture of the PiJuice board my guess is TP1 is for the pogo pin for Pi 3 and
TP3 for Pi 2

shawaj, 2017-10-27:
@tvoverbeek yes you are correct. TP1 is Pi 3, TP3 is Pi 2 or B+ or A+ and TP2 is for Pi Zero

shawaj, 2017-10-27:
[image]

as seen in this photo....

shawaj, 2017-10-27:
In fact, at some point we will have a little picture guide, telling you what all the unpopulated components, all of the test points (TP1 to TPXX), all of the unpopulated and populated headers do and everything else. Just a work in progress ATM

francesco-vannini, 2017-12-14:
@klint-k the documentation as now been updated and the pogo pin location properly documented

closed 2017-12-14 by francesco-vannini

GStainburn, 2018-12-11:
I have a Pi3 3 B+ and I have just gone to install the Pogo pin. However, none of the holes in the PiJuice board come close to matching the Run pin on the Pi 3B+.

Does this mean that the pin will not work with the 3B+?

tvoverbeek, 2018-12-11:
@GStainburn Short answer: Yes, pogo pin will not work on 3B+
The PiJuice board was designed before the 3B+ was available. Hence only support for the B+, 2B, 3B and Zero.
I do not know if PiSupply will update the board so future boards will also support the 3B+.
@shawaj ?

GStainburn, 2018-12-11:
Thank you for the quick response.  I'll use an older board.  

I must echo the previous comments about the documentation. The user guide is totally useless, and doesn't even point to better sources, which thankfully exist on-line. Fingers crossed it perform better than the PicoUPS did.

francesco-vannini, 2018-12-11:
@GStainburn the comments where referring to the documentation as it was a year ago. Since then you can find details about the pogo pins if that's what you were referring to 
https://github.com/PiSupply/PiJuice/tree/master/Hardware#unpopulated

https://user-images.githubusercontent.com/16068311/36384527-7d562e2c-1587-11e8-81d1-d78835ff8e67.png

"totally useless" feels a bit too strong don't you think?

GStainburn, 2018-12-11:
Hi

The documentation that I was referring to was the "User guide (you're reading it)".  This document does not provide any useful information on how to install or use the board.  It is very pretty and points to lots of ideas of what you can do with the Pi once the board has been installed, but is no use whatsoever in the actual install process.

The documentation on here is very good and I am working through it as time permits, and it looks like I will be able to do the project(s) I wish to do. It does look like I will be purchasing more PiJuice boards in future.

However, it would be useful if the user guide was an installation guide, or at least pointed you towards the on-line documentation

francesco-vannini, 2018-12-11:
@gstainburn did you mean this https://learn.pi-supply.com/make/pijuice-quick-start-guide-faq/#install-the-pogo-pin-connector

shawaj, 2018-12-11:
@tvoverbeek yes the future batches will all be compatible with the 3B+.

See new board design with new hole for 3B+ location.
[image]

GStainburn, 2018-12-11:
Hi
I did not look at that document because I had already received  confirmation that the PiJuice is not compatible with the 3B+ that I have here. However I will have a look at that documentation later as the target board is an older one.

My criticism is is purely with the paper user guide provided with the board.  Neither that document, or the box it came in points to the (excellent) on-line documentation.  The user guide only points to the projects pages, and the URL's on the outside of the box only take you to the shop pages which you don't need because you've already made the purchase.

Only by using Google or by reading to the bottom of the shop pages do you find a reference to the github doc's.

**** this is my only criticism. *****

A small note to add is that where in the instructions it says to remove the plastic tab, it may be an idea to say that the Pi may boot up once the tab is removed.  This is what happened with mine, but I had not connected the KVM at this point.

shawaj, 2018-12-11:
@francesco-vannini long time no see :-) glad to see your name back around here again!!

@GStainburn the guides are all linked from the product page on our website and the paper user guide does contain the links. But anyway, no issue if you did not find them as you have all the info now. Let us know if there is anything else we can help with.

francesco-vannini, 2018-12-11:
@Gstainburn no worries at all, glad you now have few more sources to look at. Hopefully I've helped you reconsider on the documentation quality.

@shawaj :)

sinistermidget, 2018-12-11:
Is there any reason why (aside from voiding warranty) that a jumper wire can't be soldered from TP1 to the Run pin on a 3B+?

GStainburn, 2018-12-12:
@sinistermidget that's a good idea.  For the moment I can just use an older Pi.

Do we know when the 3B+ compatible boards will be available? If this pilot project works I could end up wanting more (remote wildlife and security cameras) as well as robots for fun

@shawaj both links on the box  (www.pijuice.com and www.pi-supply.com) point to sales pages. Many people including me may not read the page as we've already made the purchase.  The links to the doc's are all at the bottom of those pages.  If a link to the docs was placed at the top of these pages then all would be good.

shawaj, 2018-12-12:
The new version will likely be January or February next year. 3k are in production right now.

There are more links inside the guide which go to the getting started guides

sinistermidget, 2018-12-13:
@GStainburn You may want to look at the new Pi 3A Plus for remote camera applications. The removal of Ethernet and USB hubs improves battery performance dramatically vs a 3B Plus or 3B board.

GStainburn, 2018-12-14:
@sinistermidget thanks for the info.  I think I'm going to have a move-about of my boards (Domoticz moved the new 3B+ etc) which will free up older boards for use with the PiJuice boards.  Will also consider your suggestion of soldering a jumper wire