PiSupply/PiJuice #11: @Reviewers something to test please

klint-k, 2017-10-27:
Hello

@Reviewers - could you check out this code:
`#!/usr/bin/python3

from __future__ import print_function
#import time
#import datetime
#import calendar
#import sys, getopt
from pijuice import PiJuice

pijuice = PiJuice(1,0x14)

for K in range(61):
    pijuice.status.SetLedBlink('D2', 1,(244, 66, 104), 0.3, (0, 0, 52), 0.1)
    pijuice.status.SetLedBlink('D2', 1,(255, 33, 52), 0.5, (0, 0, 0), 0.75)
`

If it works I would call it "Frankenstein" else it should be called "Igor", because Igor did not bring the right parts.

shawaj, 2017-10-27:
Thanks Klint!

klint-k, 2017-10-28:
I cant fix the indents.

shawaj, 2017-10-28:
@klint-k the reason is that you had backticks before and after each line....this doe sont allow you to do indents on GitHub.

Instead, you need to do a code block - either by highlighting and then clicking the "insert code" button at top of text entry box (looks like this <>) or you put three back ticks before the code block like ``` that and then you put another three after it like this ```

shawaj, 2017-12-14:
@francesco-vannini or @tvoverbeek reckon we could test this for @klint-k at some point? I don't actually have a PiJuice at the moment to test with (gave them all away!!!)

tvoverbeek, 2017-12-15:
@klint-k 
Sorry to say it is "Igor".
I am assuming you want the first blink to be 0.3 seconds of (244,66,104) followed by 0.1 seconds of (0,0,52) and the second one 0.5 seconds of (255,33,52) followed by 0.75 seconds of (0,0,0).
The SetLedBlink statement sends the command to the microprocessor on the PiJuice board which
executes it. Therefore you have to wait in your python program before you send the next SetLedBlink.

This code does what you intended (I hope):
```
#!/usr/bin/python
from pijuice import PiJuice
from time import sleep

pijuice = PiJuice() # default is I2C bus 1 and address 0x14
for K in range(61):
        pijuice.status.SetLedBlink('D2', 1,[244, 66, 104], 300, [0, 0, 52], 100)
        sleep(0.4)
        pijuice.status.SetLedBlink('D2', 1,[255, 33, 52], 500, [0, 0, 0], 750)
        sleep(1.25)

#  SetLedBlink notes:
#  r,g,b colors have to be given as lists: square brackets []
#  Mixed colors do not work very wel. Using only one of red, green or blue works fine.
#  Unit of period is msecs, range 0-2550
#  SetLedBlink (and SetLedState) work only for USER_LED
#  After sequence finishes, the led returns to the state before SetLedBlink
```

klint-k, 2017-12-15:
I was hoping for a heart beat.

tvoverbeek, 2017-12-15:
Well, the flashing D2 Led surely looks like a heart beat, at least for a little over a minute.

shawaj, 2018-01-24:
Is there anything else more that we need to do here?

tvoverbeek, 2018-01-25:
No, closing this issue.

closed 2018-01-25 by tvoverbeek