PiSupply/PiJuice #22: Add description of manual installs

shawaj, 2017-12-14:
Add description of a manual package install just in case and also of a local compile etc if necessary

shawaj, 2017-12-14:
FYI @francesco-vannini  - some notes:

installing on raspbian-lite that is headless by default will automatically install python-tk and python-gtk dependencies.

If it is using some special distribution it is possible to build package without config gui and status tray and one way to do this is to remove pijuice-gui.desktop, pijuice-tray.desktop from data directory in sources/. Also it needs to remove python-tk, python-gtk2 from dependencies line in sources/debian/control file.

Automatic shutdown after power source disappears and than automatic boot when power source appears. There is no special feature to handle that case. One way to mimic this it to set minimum charge or minimum battery voltage in system task, than select SYS_FUNC_HALT_POW_OFF function for low charge or low battery voltage in System Events tab. For automatic power on can be used Wakeup on charge trigger.

shawaj, 2018-01-24:
@francesco-vannini looks like you have sorted most of this already here - https://github.com/PiSupply/PiJuice/tree/master/Software

And I think the rest is handled in the recent firmware / software updates by the looks of it?

Are you happy for this to be closed?

francesco-vannini, 2018-01-25:
Sure, yes

closed 2018-01-25 by francesco-vannini