PiSupply/PiJuice #29: potential to reduce CPU clock speed to increase battery life?

shawaj, 2017-12-29:
@francesco-vannini was inspired by the recent news about apple dynamically reducing the CPU power on their devices due to aging batteries - and had a neat idea.

Potential to underclock the CPU to increase battery life.

Needs further investigation

klint-k, 2018-01-06:
Only a good idea if it has some use control.

shawaj, 2018-01-06:
@klint-k what do you mean?

klint-k, 2018-01-07:
The user needs to have the choice, to slow down or not.

shawaj, 2018-01-07:
Yep, that was the whole idea

shawaj, 2018-01-12:
https://raspberrypi.stackexchange.com/questions/8877/overclocking-via-command-line

And

http://www.pantz.org/software/cpufreq/usingcpufreqonlinux.html

And

http://www.jackenhack.com/raspberry-pi-3-overclocking/

And

https://www.raspberrypi.org/forums/viewtopic.php?t=138123

And

https://www.raspberrypi.org/forums/viewtopic.php?t=33159

And

https://www.raspberrypi.org/forums/viewtopic.php?t=19253

shawaj, 2018-01-12:
Basically, it seems that he underclock and under volting will provide savings and is very easy to do (just a config file edit and reboot so we could automate this) but the power savings may end up being negligible in comparison to the power consumed by LAN9514 chip and any USB peripherals.

Worth playing around with and testing though. Although it seems best option for lower power consumption is model A or Zero v1.3

Maybe we should do an article called "extreme pi power saving" or something ;-)

ChristopherRush, 2019-03-14:
Had a look at this and came up with the following guide - https://learn.pi-supply.com/make/how-to-save-power-on-your-raspberry-pi/

The power savings from underclocking/undervolting where actual minimal. The biggest saving was to disable the LAN chip, which almost doubled the battery life when IDLE.

francesco-vannini, 2019-03-15:
I agree with frequency scaling the CPU already uses very little when IDLE.
As for the system LEDs there are differences between the Zero and the A/B ones
[https://www.jeffgeerling.com/blogs/jeff-geerling/controlling-pwr-act-leds-raspberry-pi](url)
and I find confusing the logic between the two a part from one having one LED and the other having two.
In  my script I use this at the start of the service:
```
revision=`cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}'`

if [[ "9000" == ${revision:0:4} ]]; then
    echo 1 | sudo tee /sys/class/leds/led0/brightness
else
    echo 0 | sudo tee /sys/class/leds/led0/brightness
    echo 0 | sudo tee /sys/class/leds/led1/brightness
fi
```
which determines if the Pi is a Zero or not and acts on the LEDs accordingly.