PiSupply/PiJuice #31: Created a user script to run once SW2 is pressed

lesp, 2018-01-10:
I've created a user script that I wish to run when SW2 is pressed. I've added the path to the script to the user scripts tab and configured HAT to trigger that script when pressed...but zilch.
Also when I close the PiJuice Configuration app it reverts the user script back to a previous script that I was testing.

The file /var/lib/pijuice/pijuice_config.JSON exists. 
Permissions for the file
-rw-r--r--  1 pi   pi     77 Jan  5 13:31 pijuice_config.JSON

Contents of pijuice_config.JSON (Which shows the ORIGINAL command I entered in the user func)
{
  "user_functions": {
    "USER_FUNC1": "idle3"
  }, 
  "system_task": {}
}
Thanks!

shawaj, 2018-01-10:
@lesp not that it will necessarily make any difference, but what were the scripts you were trying to add in there? And do you know what permissions they had?

Just trying to see if we can replicated the issue.

lesp, 2018-01-10:
The first attempt was to call idle3 when SW2 was pressed, this is the Python 3 editor in /usr/bin

==Details==
pi@raspberrypi:/usr/bin $ ls -lha | grep idle3
-rwxr-xr-x  1 root root       92 Jan 20  2017 idle3
==END==

The second script was something I whipped up called les.sh (original) in /home/pi/
==Details==
-rwxr-xr-x  1 pi   pi     23 Jan  5 14:15 les.sh
==END==

And that contained a simple bash script to open idle3 and run in the background.

==Contents of les.sh==
#! /bin/bash

idle3 &
==END==

francesco-vannini, 2018-01-10:
Try
`sudo chmod 777 /var/lib/pijuice/pijuice_config.JSON`
that won't solve your problems with the SW not running your script but I'd like to see if it helps changing the content of the JSON file hence reflecting that in the GUI.

francesco-vannini, 2018-01-10:
Could you also provide the output of
`systemctl status pijuice.service`

lesp, 2018-01-10:
chmod has changed to make the file rwx for all, but when changing contents of user func 1, and then closing the app, once re-opened the GUI user func 1 shows old user script, this is confirmed in the JSON file.

==systemctl==
pi@raspberrypi:~ $ systemctl status pijuice.service
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-01-10 15:53:10 UTC; 1h 29min ago
  Process: 809 ExecStopPost=/usr/bin/pijuice_sys.py stop (code=exited, status=0/SUCCESS)
  Process: 803 ExecStart=/usr/bin/pijuice_sys.py (code=exited, status=1/FAILURE)
 Main PID: 803 (code=exited, status=1/FAILURE)

Jan 10 15:53:10 raspberrypi systemd[1]: pijuice.service: Failed with result 'exit-code'.
Jan 10 15:53:10 raspberrypi systemd[1]: pijuice.service: Service hold-off time over, scheduling restart.
Jan 10 15:53:10 raspberrypi systemd[1]: Stopped PiJuice status service.
Jan 10 15:53:10 raspberrypi systemd[1]: pijuice.service: Start request repeated too quickly.
Jan 10 15:53:10 raspberrypi systemd[1]: Failed to start PiJuice status service.
Jan 10 15:53:10 raspberrypi systemd[1]: pijuice.service: Unit entered failed state.
Jan 10 15:53:10 raspberrypi systemd[1]: pijuice.service: Failed with result 'exit-code'.

==END==

tvoverbeek, 2018-01-10:
Normally the `pijuice_config.JSON` file is owned by user pi, group pi since the python gui app is run as user pi when started from the Desktop.
The directory `/var/lib/pijuice` should have protection `rwxrwxrwx` otherwise the python gui cannot write/update the pijuice_config.JSON file.
The service (`/usr/bin/pijuice_sys.py`) exits immediately when it cannot access/open the JSON file.
This seems to be what happened looking at the output from `systemctl status pijuice.service`
Also your les.sh file will execute but idle3 will abort with an error: `no display name and no $DISPLAY environment variable` since it is started by the service under the root user.
Modify les.sh as follows:
```
#!/bin/bash

export DISPLAY=:0.0
idle3
```
Works on my system
Hope this helps

shawaj, 2018-01-10:
@tvoverbeek is there a better way to handle this all do you think, so that it kind of "works out of the box" a bit better? Or maybe it is just a case of documenting the process. My brain is a bit muddled today I can't think of a good way to do it.

francesco-vannini, 2018-01-12:
@lesp did any of the above work?

lesp, 2018-01-12:
Sadly not, I even changed the owner of the pijuice directory to pi:pi 
Service is still failing :(

tvoverbeek, 2018-01-12:
Try to run the service program /usr/bin/pijuice_sys.py directly:
```
# Ensure service is not running or restarting
sudo systemctl disable pijuice.service
sudo systemctl stop pijuice.service
# Check
systemctl status pijuice.service
# Run directly in terminal
sudo /usr/bin/pijuice_sys.py
```

This produces more output and hopefully a reason why the service does not start

lesp, 2018-01-12:
@tvoverbeek Steps followed and this is the output of running pijuice_sys.py

===OUTPUT===

pi@raspberrypi:~ $ sudo /usr/bin/pijuice_sys.py
Traceback (most recent call last):
  File "/usr/bin/pijuice_sys.py", line 253, in <module>
    main()
  File "/usr/bin/pijuice_sys.py", line 229, in main
    if configData['system_task']['enabled']:
KeyError: 'enabled'

===END===

tvoverbeek, 2018-01-12:
Smells like an incomplete `/var/lib/pijuice/pijuice_config.JSON`
A minimal `/var/lib/pijuice/pijuice_config.JSON` should look like:
```
{
   "system_task": {
        "enabled": false
   }
}
```
The service expects the "enabled" key for "system_task" to be present in `/var/lib/pijuice/pijuice_config.JSON`

How does your `/var/lib/pijuice/pijuice_config.JSON` look like?
In your case the "enabled" key for "system_task" is obviously missing.

Note that when system_task is not enabled the service exits after some initial checks.

shawaj, 2018-01-24:
@tvoverbeek @francesco-vannini @DenisKhrutsky is there anything else you think we need to do here?

@lesp - have you still got the PiJuice? We have done a lot of updates to the software recently, and if you still have it it would be great to get your feedback, if any. Although appreciate you may not have time :-)

PGFleckers, 2018-03-25:
I have tried to make use of SW also to run a simple python script, I have tried running the script directly and also from within a .sh but nothing

tvoverbeek, 2018-03-26:
@PGFleckers This works with the last software version.
(At least it works for me).

Some things to check:
- System task has to be enabled
- Check Button configuration on the Buttons tab of the 'PiJuice HAT Configuration' window
- Check assignment of USER_FUNCn to your python functions on the 'User Scripts' tab of the 'PiJuice Settings' window.
- Your python scripts have to be executable unless you specify 'python myscript.py'  instead of 'myscript.py' in the 'User Scripts' tab.
- Since your script will be executed by the pijuice service it runs as user root, not pi.
- For debugging this have a look at user_func1.py in [https://github.com/PiSupply/PiJuice/blob/master/Software/Test/user_func1.py](url)

PGFleckers, 2018-03-26:
Carried out an upgrade and SW now runs script successfully, thanks.

shawaj, 2018-05-18:
@ChristopherRush @tvoverbeek if I am correct, this now works correctly? Can we close do you think?

shawaj, 2018-05-18:
Or perhaps we need more documentation to explain this?

tvoverbeek, 2018-05-18:
Scripts started by the service run as root. Should be documented somewhere.

shawaj, 2018-05-18:
@ChristopherRush can you document this?

ChristopherRush, 2018-05-21:
@shawaj Documented

closed 2018-05-21 by ChristopherRush

Ruben138, 2025-02-24:
Yo

Ruben138, 2025-02-24:
Yo