PiSupply/PiJuice #32: Picture of switches and LEDs in main README is wrong

tvoverbeek, 2018-01-10:
In the main README -> Basic Functionality -> Power On and Off the Pi the picture shows the LEDs and switches in the order SW2 - LED2 - SW3 - SW1 - LED1.
The LED1 and LED2 labels have to be exchanged.
The picture in the hardware section is correct.
Also I had trouble orienting myself with the picture in the main README, since it is not
clear where the edge of the board is located (bottom of the picture) especially since
LED1 and LED2 labels are in the wrong place.

francesco-vannini, 2018-01-10:
The picture in the main README is simply a cutout of the one in the hardware section. The sequence of switches and LEDs from left to right looking at the board from the top is the one indicated. Am I missing something?

tvoverbeek, 2018-01-10:
OK, pictures are the same, but in the GUI the charge-status LED is D1 (=LED2 in the picture) and the user led is D2 (=LED1 in the picture). Not very consistent.
Would still suggest to make the cut-out from the hardware section a bit bigger, so the edge is visible. See attached.
[image]

shawaj, 2018-01-10:
Hmm - if you close up on the picture you posted @tvoverbeek  LED2 is labelled as D2 and LED1 is labelled D1. Perhaps they are mislabeled in the software rather than actually here by @francesco-vannini ?

Ton - did you test this in the GUI - i.e. changing settings for D1 in GUI changes settings for D2/LED2 on the board?

shawaj, 2018-01-10:
[image]
better picture here

francesco-vannini, 2018-01-10:
Could it not be that you have an old version of the firmware? There was an Led swap at some point.

shawaj, 2018-01-10:
Hmm - possibly yes. Forgot about that. Although I would be surprised if Tons was one with the old firmware but I suppose it is possible.

@francesco-vannini is this the latest firmware here? https://github.com/PiSupply/PiJuice/tree/master/Firmware

shawaj, 2018-01-10:
@francesco-vannini just thinking - I guess we should do a changelog of the firmware (and the software) - what do you think? saw you were implementing something similar on PaPiRus?

francesco-vannini, 2018-01-11:
We have defined a versioning system and yes we should add some release notes or a change log with that

tvoverbeek, 2018-01-11:
If I have the old firmware, can I check this, e.g. with GetFirmwareVersion()?
How do I update to the current one, since I do not have the old firmware updater?
There were notes that you have to use the old firmware updater to get the current version.
Anyway, the function of the LEDs is definitely swapped on my PiJuice.

francesco-vannini, 2018-01-11:
You can follow the instructions in the Firmware section as yours won't be that old
https://github.com/PiSupply/PiJuice/tree/master/Firmware
use the latest firmware in that folder
simply run
`./pijuiceboot 14 PiJuice-V1.0_2017_10_23.elf.binary`

Versioning of the firmware with change logs is also something @mmilann will provide as of  the next release so that at least from the GUI in the Firmware tab people will be able to tell which they have.

shawaj, 2018-01-11:
And we should also be able to update the firmware from the GUI as well? Or does that not work currently @francesco-vannini ?

Also are the LEDs the right way around on yours. I have no PiJuice ATM so can't check!

francesco-vannini, 2018-01-11:
My version is OK I suspect all boards send out so far have V1.0_2017_08_15. As you can see below the problem was fixed in September. The trouble with the current firmware is that the version number wasn't updated so there is no easy way to tell which one people have. The good news is that the procedure to update the firmware manually which is described in the Firmware section is fine because V1.0_2017_08_15 is recent enough and people like @tvoverbeek  can simply update to the latest V1.0_2017_10_23. As of the next one the versioning should start increasing according to what we said #33 
```text
V1.0_2017_10_23
Added command to configure charging, enable/disable
Added command to configure inputs, disable/enable 5V io input to charger, set usb micro in current limit, set usb micro in DPM voltage
V1.0_2017_09_25
LEDs D1 and D2 swapped to match labels on boards.
```

As for the GUI update I described the work around currently needed as we do not yet provide a "Browse to" function. https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-config-firmware-menu
We have already documented this as an improvement.

shawaj, 2018-01-11:
Where was that changelog @francesco-vannini - was that something you developed with Milan? Hadnt seen that yet!

Aha - yes - I remember now about the "browse to" - thanks for reminding me!

shawaj, 2018-01-11:
@tvoverbeek can you update firmware and see if that fixes the issue?

francesco-vannini, 2018-01-11:
@shawaj just took information from google drive

tvoverbeek, 2018-01-11:
Updating with PiJuice-V1.0_2017_10_23.elf.binary switches the LEDs the 'right' way.
Note the firmware in the current raspbian pijuice package has size 64600 and date June 5, 2017
and has the LEDs switched.
So update the official package asap, otherwise you get problems like I had after shipping the first 2K units.

closed 2018-01-11 by tvoverbeek

francesco-vannini, 2018-01-11:
Thanks @tvoverbeek we should be having a new release of software and firmware by next week. The boards will ship with the firmware you just updated to I'd imagine @shawaj do you know? But absolutely you are right we will test the new code and get it on the official APT repo ASAP

shawaj, 2018-01-12:
@francesco-vannini @tvoverbeek yes, the firmware programming will happen independent of the PiJuice software so will use the current one regardless. Your board must have been one of the first batch that has the old firmware.

But I'll triple check with Embest

tvoverbeek, 2018-01-12:
I do not know what the original version on the board was.
During initial testing I tried the upload firmware in the gui and that uploaded the firmware in the raspbian package, i.e. the old firmware.
One more reason to update the raspbian package soon.

shawaj, 2018-01-12:
Spoke to Embest today. They're shipping the first 2k with the October firmware because the programming of the chips is already underway.

So it's definitely important to have the new firmware release from Milan included in the software package and/or a "browse to" capability. As we will need them to update the firmware on first use to get the new features etc.

francesco-vannini, 2018-01-15:
@DenisKhrutsky is going to provide a "Browse to" hopefully even in this new release.
The firmware in the deb package has anyway been updated to the latest one

shawaj, 2018-01-15:
Awesome. The latest debs are ready with names pijuice and pijuice-lite yeah?

And can be submitted to Raspbian?

shawaj, 2018-01-15:
@DenisKhrutsky @francesco-vannini can the PiJuice-lite and PiJuice packages be installed alongside each other without issues?

francesco-vannini, 2018-01-15:
pijuice and pijuice-light (not lite). As I was trying to tell you on Whatsapp we've done more or less all we could without the firmware, we'll start testing as soon as we receive it then focus on bug fixing and uploading a stable version only of the GUI software to be sent to Raspbian. The light for the moment will have to be obtainable from Github.
I imagine they can be installed alongside each other but I am not sure

shawaj, 2018-01-15:
Hmm. Not following.

- Why wouldn't we put it in the repo now? This isn't a good idea IMO. It needs to be easy to install and not require GitHub. Before we ship really.

- they need to be able to be installed side by side so it needs to be PiJuice light establishes core tools and PiJuice adds GUI stuff and has PiJuice light as a dependency, or it needs to have some other way of working together

- we need a version with updated firmware on the Raspbian repo ASAP. Because people will need to do a firmware update immediately on receipt of the device.

- we also have some in the wild ATM that could do with the updates ASAP, for reviews and such

Also already got the updated firmware...

francesco-vannini, 2018-01-15:
I said that the full (GUI) version is going to Raspbian and that it has the latest firmware. With this new firmware we will first have to test now, implement the additional changes @mmilann  as done to the software also in the dev branch then repackage all and get it ready to be sent.
The light version has some problems to be sent out now, @DenisKhrutsky can explain further so I suggest that for this release we might have the full via APT and the light, which is anyway for advanced users, via Github. This can all be changed after this release.
@DenisKhrutsky can you clarify if the light and full can be installed side by side?

tvoverbeek, 2018-01-15:
As far as I know the light version is a subset of the full (GUI) one and therefore cannot be installed side by side since they have common files. (a file can only belong to a single package)
The only way would be if the GUI one has the light one as dependency, and the GUI one does not contain the files from the light one (i.e. pijuice.py, pijuice_sys.py)
Moreover a light one only makes sense if it includes a text based config utility (curses based copy of the GUI one?) otherwise only the default settings will work.
Just my 5 cents.

shawaj, 2018-01-15:
Yeah I think that's the best way to do it @tvoverbeek - I think we definitely want to have it so that you can use them side by side and I think it makes the most sense to do that by having pijuice-lite as the core and building on top of that but maybe @DenisKhrutsky has a better way

Also Ton, everything is configurable through the JSON file so a curses menu isn't necessary but would definitely be useful

But certainly ability to update firmware and stuff via the lite one is useful.

@francesco-vannini it would be good to have both sorted before launch the 23rd ideally

DenisKhrutsky, 2018-01-15:
@tvoverbeek is right, pijuice and pijuice-light can't be installed simultaneously. In the future it's nice to have two separate packages as mentioned before: pijuice for core functionality and pijuice-gui for, well, GUI. But that needs some time to establish and test. Updating firmware via console is definitely an important feature, I'm going to work on it.

shawaj, 2018-01-15:
@DenisKhrutsky the problem is that we need people to have the use of the PiJuice light now... Some industrial customers. What sort of timeline do you think it would require to do this for developing and testing?

DenisKhrutsky, 2018-01-15:
@shawaj If we need to give this task a top priority, it can be done near 19th

shawaj, 2018-01-15:
@DenisKhrutsky thanks very much that would be great

shawaj, 2018-01-15:
@francesco-vannini yes but we need something up on Raspbian and it needs to be compatible with future

tvoverbeek, 2018-01-15:
For now bring the existing pijuice package (including GUI) up to date.
You can specify in the pijuice-light control file that it conflicts with pijuice.
Later we can supersede it with a pijuice-base and pijuice-gui package.
See chapter 7 in the debian policy manual:
https://www.debian.org/doc/debian-policy/#document-ch-relationships
especially sections 7.3, 7.4 and 7.6

DenisKhrutsky, 2018-01-15:
@tvoverbeek Thank you very much for directions.

tvoverbeek, 2018-01-16:
@shawaj Yes, you can change the package name if you specify that it conflicts with the previous package name.
You can also keep the name and still specify it conflicts with the same package for versions < given version.
Suggest to first get the existing (full) pijuice package in a stable (= iron out the bugs) version before we go ahead with the split.
For your industrial customers provide the light package and specify in its control file that it conflicts with the full pijuice.
You probably will have to supply them with a pre-configured JSON settings file, since there is no config utility in the light package.

shawaj, 2018-01-16:
Hmm. Agree with you, but not ideal because of the customer's requirements. But it sounds like Denis can sort this all pretty quickly so it's not a massive issue. We can wait until the end of the month to iron out now bugs.

Also sent @DenisKhrutsky a PiJuice so easier to debug as we will all have one then,