PiSupply/PiJuice #33: Software/Firmware versioning

francesco-vannini, 2018-01-11:
We have decided to adopt a two digit group versioning with major version for big group of fixes and minor for each build.
We should adopt the same for both software and firmware.

Additionally we need to provide a change log for SW and FW in a descriptive form listing bugs and improvements

The change log shall be one file only with two sections:
- Header Section explaining how to check the versions from command line or GUI, other info
- Version x.y
- Software
Improvements
Fixes
- Firmware
Improvements
Fixes
-Append of the previous Versions

Next software release will be 1.1 and so will it be for firmware I reckon.

@DenisKhrutsky @mmilann @tvoverbeek @shawaj any thoughts on this are welcome.

shawaj, 2018-01-11:
Seems like a good way to do things to me :-)

tvoverbeek, 2018-01-11:
Looks good to me.

francesco-vannini, 2018-01-19:
The agree format for the Firmware filename is:
PiJuice-V<major_version>.<minor_version>_<date>.elf.binary
E.g. PiJuice-V1.1_2018_01_15.elf.binary
@DenisKhrutsky already implemented this logic in pijuice_gui.fw for the Firmware tab

This has also been discussed in  https://github.com/PiSupply/PiJuice/issues/34

francesco-vannini, 2018-01-19:
Sure, what would the format look like then? I have troubles figuring that out

mmilann, 2018-01-19:
For example variantname_V<major_version>.<minor_version>_.elf.binary.

And then have it decoded:
variant_names = [PiJuice, PijuiceVariant1...]

shawaj, 2018-05-18:
https://github.com/PiSupply/PiJuice/blob/master/CHANGELOG.md now details the changelog and the versioning is accepted. I will move this discussion / explanation of the versioning over to there

closed 2018-05-18 by shawaj