PiSupply/PiJuice #34: GetFirmwareVersion() returns wrong(?) result

tvoverbeek, 2018-01-11:
PiJuice.config.GetFirmvareVersion() now returns version='10', variant='0'.
With the agreed versioning I would expect version='1', variant='0'.
For the upcoming release version='1', variant='1'
Also note the current spelling of GetFirm**v**areVersion (v instead of w).
Intentional? Or just a typo?

shawaj, 2018-01-12:
Think that's probably a typo yes and also agreed about the versioning

tvoverbeek, 2018-01-12:
Note, the typo occurs both in pijuice.py and pijuice_gui.py

francesco-vannini, 2018-01-12:
@mmilann could you please fix the version number according to what we decided here
https://github.com/PiSupply/PiJuice/issues/33
The release you are about to send out should therefore be 1.1
Would that be ok?

@tvoverbeek @DenisKhrutsky already refactor the method name in the dev branch, thanks for pointing this out

francesco-vannini, 2018-01-18:
With new firmware V1.1_2018_01_15
```text
pijuice.config.GetFirmwareVersion()
{'data': {'version': '11', 'variant': '0'}, 'error': 'NO_ERROR'}
```
whereas we should get 
```text
{'data': {'version': '1', 'variant': '1'}, 'error': 'NO_ERROR'}
```

francesco-vannini, 2018-01-18:
@tvoverbeek suggested:
Expects GetFirmwareVersion for existing version to return a 2 digit string
    as version x.y. Variant field ignored.
    Also new version in file name can only be a 2 digit string (x.y).
    => Max major version is 9
    => Update check needs rethinking/redesign and include the variant field.

Also the files we are adding to the repository are in the form  PiJuice-V1.1_2018_01_15.elf.binary and  PiJuice-V1.1_2018_01_15.elf.hex @DenisKhrutsky software logic should check the current version on that type of naming convention.

mmilann, 2018-01-18:
Firmware version is represented by one byte or two hex digits, so there are 16 possibilities for minor digit and 16 for major. Firmware variant is about to distinguish firmware variants for different PiJuice hardware variants in future. So for different versions of firmware variant is same, '0'.

closed 2018-01-18 by mmilann

francesco-vannini, 2018-01-18:
Thanks for clarifying that @mmilann

shawaj, 2018-01-18:
@mmilann is there a way to add more versions than 16 in future if necessary?

Doubt we will need to but just a thought

tvoverbeek, 2018-01-18:
Using the major (0-9) and minor (0-9) fields you have theoretically 100 versions (unless you go to Hexadecimal).
We are already at 1.1 => 88 versions left.

shawaj, 2018-01-18:
Probably won't ever get to 88 but may as well use hexadecimal too as firmware is rarely going to be changed so doesn't need to be particularly human friendly

shawaj, 2018-01-18:
Thanks @mmilann probably ok for now anyway