PiSupply/PiJuice #35: Test script pijuice.sh does not work.

tvoverbeek, 2018-01-11:
The pijuice.sh bash script in Software/Test does not work at all (wrong commands).
Below a modified version which works.
Should be checked by somebody knowing the firmware.
```
#!/bin/bash
#
# This script assumes the pijuice service is not running.
# Otherwise the long press on SW1 will shut down the Pi.
#

BUS=1
ID=0x14

GET_BUT_EVT_CMD="BUT_EVT=\$(i2cget -y $BUS $ID \0x45  b |sed -e "s/^0x//" |tr "a-z" "A-Z")"
GET_RSOC_CMD="RSOC=\$(i2cget -y $BUS $ID \0x41  b |sed -e "s/^0x//" |tr "a-z" "A-Z")"
GET_VER_CMD="VER=\$(i2cget -y $BUS $ID \0xFD  w |sed -e "s/^0x//" |tr "a-z" "A-Z")"
GET_CUR_LOW_CMD="CUR_LOW=\$(i2cget -y $BUS $ID \0x4f  b |sed -e "s/^0x//" |tr "a-z" "A-Z")"
GET_CUR_HIGH_CMD="CUR_HIGH=\$(i2cget -y $BUS $ID \0x50  b |sed -e "s/^0x//" |tr "a-z" "A-Z")"
GET_TEMP_LOW_CMD="TEMP_LOW=\$(i2cget -y $BUS $ID \0x47  b |sed -e "s/^0x//" |tr "a-z" "A-Z")"
GET_TEMP_HIGH_CMD="TEMP_HIGH=\$(i2cget -y $BUS $ID \0x48  b |sed -e "s/^0x//" |tr "a-z" "A-Z")"
GET_BATVOLT_LOW_CMD="BATVOLT_LOW=\$(i2cget -y $BUS $ID \0x49  b |sed -e "s/^0x//" |tr "a-z" "A-Z")"
GET_BATVOLT_HIGH_CMD="BATVOLT_HIGH=\$(i2cget -y $BUS $ID \0x4A  b |sed -e "s/^0x//" |tr "a-z" "A-Z")"

eval $GET_VER_CMD
echo Firmvare version: V${VER:2:2}.${VER:0:2}

# Clear button 1  events
i2cset -y $BUS $ID 0x45 0xf0 0xff 0xf0 i

while true; do

eval $GET_BUT_EVT_CMD
eval $GET_RSOC_CMD
eval $GET_CUR_LOW_CMD
eval $GET_CUR_HIGH_CMD
eval $GET_TEMP_LOW_CMD
eval $GET_TEMP_HIGH_CMD
eval $GET_BATVOLT_LOW_CMD
eval $GET_BATVOLT_HIGH_CMD

echo CHARGE = $((0x${RSOC}))%, Temp = $((0x${TEMP_HIGH}${TEMP_LOW}))[10K], VBAT = $((0x${BATVOLT_HIGH}${BATVOLT_LOW})) mV, #Current = $((0x${CUR_HIGH}${CUR_LOW})) mA,

if [ $BUT_EVT = "05" ]; then
    # Long press (10sec) on SW1
    break
fi

sleep 1
done 
```

mmilann, 2018-01-11:
This script is compatible only to very first version of firmware and is outdated. New firmware supports only  I2C block read/write.

shawaj, 2018-01-12:
Is it worth updating to this method?

mmilann, 2018-01-12:
This method is not efficient and will be too much overload to support it.

tvoverbeek, 2018-01-12:
Suggest to delete it and clean up pijuicetest.py instead.
Also V1.1.txt needs to be deleted (=fragment of a changelog)

DenisKhrutsky, 2018-01-15:
pijuice.sh and V1.1.txt have been deleted. Improvements to pijuicetest.py should be discussed later.

shawaj, 2018-01-15:
Thanks @DenisKhrutsky let's leave this then and discuss updates later

shawaj, 2018-05-18:
@tvoverbeek do you think we can close this issue or should we ask @wikipedia555 to perform the updates that @DenisKhrutsky suggested?

tvoverbeek, 2018-05-18:
We should keep pijuicetest.py, but it needs some cleaning up.
It seems to test most of the functions of pijuice.py.