PiSupply/PiJuice #36: Debian packages issues on development branch

tvoverbeek, 2018-01-12:
For version 1.1 the package has changed name from pijuice to python-pijuice.
Since the pijuice package is already in the raspbian archives, this might cause problems.
For instance the prerm script should now also remove existing installs of pijuice.
control should indicate this replaces the pijuice package.

Both the -light (still on 1.0) and the python-pijuice package contain old firmware file PiJuice.elf.binary with size 64600

tvoverbeek, 2018-01-12:
Maybe I should shut up for a while and let Denis do his work ;-)

francesco-vannini, 2018-01-12:
Not at all @tvoverbeek we didn't spot that, keep them coming

DenisKhrutsky, 2018-01-12:
I missed that one. Version deployed on github was only for purpose of testing the install process, correct version is getting ready right now (with proper name and changelog).

francesco-vannini, 2018-01-12:
Also moved the most recent firmware under Software/Source/data/firmware

shawaj, 2018-01-12:
So the process will be to have pijuice package and then a pijuice-lite also?

Also @DenisKhrutsky I have got a PiJuice board ready to send you now. Will get it sent over to you next week.

DenisKhrutsky, 2018-01-15:
The latest build contains the new firmware and the problems you mentioned have been fixed.

shawaj, 2018-01-15:
Where is the latest build?

francesco-vannini, 2018-01-15:
@shawaj https://github.com/PiSupply/PiJuice/tree/develop/Software/Install

francesco-vannini, 2018-01-24:
@tvoverbeek are you happy to close this?

tvoverbeek, 2018-01-24:
Yes, this one can be closed.
Tried to build the new 1.1 debs (now on master branch) using the pckg-pijuice.sh script.
The base deb builds fine but the gui deb build fails because version is not defined.
I think I know what the problem is, will do a bit more testing and report.
What I want to test at the end is the proper reaction when trying to isntall the 1.1 deb when pijuice-1.0 is installed

francesco-vannini, 2018-01-24:
Thanks we will do a bit more testing too between today and tomorrow and very likely we'll need to re-upload to the repo. So please whatever you find out is absolutely welcome

tvoverbeek, 2018-01-24:
See #58

closed 2018-01-25 by francesco-vannini