PiSupply/PiJuice #37: Cooperation between the service (pyjuice_sys.py) and the gui (pi_juice_gui.py) missing

tvoverbeek, 2018-01-12:
The service only reads the configuration when it starts.
When initially the system_task is disabled the service exits after the initial checks.
If the system_task is active any changes in the gui for the service are not seen by pijuice_sys.py.
Also when enabling the system_task in the gui the service is not (re)started.

This could be confusing for new users, unless you document clearly that a restart is required
after changing settings in the gui.

shawaj, 2018-01-12:
Good spot. Think it make sense to do all the restarting of services automatically and stuff if possible.... Or perhaps do it on a prompt "do you want to restart now or later" type thing if it *has* to be a full on/off restart not just the services

francesco-vannini, 2018-01-19:
Moved to #48