PiSupply/PiJuice #38: Enable/Disable charging, How?? Any command to force reset/restart of the firmware?

tvoverbeek, 2018-01-12:
In the changelog for the 2017-10-23 firmware it says:
```
Added command to configure charging, enable/disable
Added command to configure inputs, disable/enable 5V io input to charger, set usb micro in current limit, set usb micro in DPM voltage
```
Are these new commands already supported in pijuice.py?

Anyway I managed to get my PiJuice in a state where it was not charging the battery, with the power connected to the 5V PiJuice USB.
Power went to the Pi, but the battery remained at constant charge level (no blue blinking LED1).
Got it back to charging by shutting down the Pi, removing external power, removing the battery and reinsert the battery, reconnect power to PiJuice USB.

shawaj, 2018-05-18:
@tvoverbeek I think the recent updates have fixed this, if I am correct?

tvoverbeek, 2018-05-18:
Yes. Closing this one

closed 2018-05-18 by tvoverbeek