PiSupply/PiJuice #39: System Switch, How does it work? Not settable from HAT tab

tvoverbeek, 2018-01-12:
I have no idea what the System Switch referenced in the HAT tab of the GUI does.
Moreover you cannot change it in the HAT tab (unless you are very very fast) because _RefreshStatus() overwrites the values every second.

OK. This is the last one for today

shawaj, 2018-01-12:
If I remember correctly the system switch state setting is to change the Vsys output max current capability to 0 (off), 500ma and 2100ma, or something like that. I thought we had already documented this somewhere but can't remember where.

@francesco-vannini do you remember what this is for / have you got it on any notes from Milan?

francesco-vannini, 2018-01-15:
No I don't know, I have added it to the questions document

shawaj, 2018-01-15:
Notified Milan too?

francesco-vannini, 2018-01-15:
He says: "I have updated this to be easier to switch with radio buttons." also this is what we have on the PiJuice Questions document:
VSYS on P3 is same as VSYS on J3 and is switchable battery voltage for system use, like connecting to Pibot power input. VSYS switch is programmable with software with “OFF”, “ON 500mA current limit” and “ON 2100mA current limit”. 
There is no Vcc label on P3 header. 5V connects to RPI_5V GPIO and current capacity depends on battery capacity, available current is shared with RPI. For BP7X if RPI consumes ~500mA than there is around 800mA left to take from this pin. 3V3 on P3 is sourced with internal LDO and maximum available to use externally is 100mA.

shawaj, 2018-01-15:
Yep so it's for vsys output rather than the regulated 5v for Pi

shawaj, 2018-05-18:
@ChristopherRush do you know if this is documented anywhere? I think we can probably close this issue but not sure what you think.

ChristopherRush, 2018-05-18:
@shawaj You can close this, it is documented under Hardware and briefly under Software.

shawaj, 2018-05-18:
thought so! just wanted a second opinion :-)

closed 2018-05-18 by shawaj