PiSupply/PiJuice #40: Watchdog misleading information (?)

tvoverbeek, 2018-01-14:
The watchdog description in the READMEs implies it watches the RPi.
However there is no watchdog reset command sent by the service on the RPi.
The watchdog timer is part of the PiJuice MCU (STM32) and only resets the MCU when it times out (as far as I know).
How was this supposed to work with the RPi?

shawaj, 2018-01-14:
It watches a (configurable) heartbeat from the PiJuice software and if it doesn't get it within the set time period it resets the pi

tvoverbeek, 2018-01-15:
That is how I expect it to work, but it does **not** as far as I can tell.

For that the RPi has to send its heartbeat to the STM32. If the STM32 does not get it within the configurable timeout the STM32 resets the RPi.
The most obvious candidate to send the heartbeat is the service pijuice_sys.py.
However it only sends the watchdog timer setting (if configured) when it starts.
The system_task loop does not send anything (i.e. reset the watchdog timer)
So again, how is this working with the current implementation?
I have no insight in the firmware. This is based on looking at the pijuice_sys and pijuice sources.

mmilann, 2018-01-15:
There is no special heartbeat call, but it is reading status in system task loop:
pijuice.status.GetStatus()

shawaj, 2018-01-15:
@mmilan so how does the PiJuice know if the RasPi has crashed?

shawaj, 2018-01-15:
Oops that should have said @mmilann

tvoverbeek, 2018-01-15:
@mmilann OK. Will test it by disabling the GetStatus() and see if it resets after 2 minutes.

mmilann, 2018-01-15:
Basically it watches of any command received with defined timeout. Status command is just most suitable in system task. One test could be to block I2C line, for example short SDA to GND, that will prevent mcu to receive any command.

shawaj, 2018-01-15:
As long as it is a reliable method, I think anything is fine really

tvoverbeek, 2018-01-15:
TL;DR Watchdog reset works

This is what I did:
- Commented out the disable watchdog when the service is stopped.
- Killed the tray app (no more get chargelevel commands)
- Started the service with systemctl
- After a while stopped the service with systemctl
- A little over 2 minutes later the RPi reset (Watchdog timeout setting was 2 minutes)

closed 2018-01-15 by tvoverbeek