PiSupply/PiJuice #41: Low charge and low battery voltage detection do not work

tvoverbeek, 2018-01-14:
I set up a user_func1 which logs events as described in the Software->Test README.
Put the limit for low charge at 15%.
Started with a full battery and disconnected it from the USB power.
Button events (SW2 triggering USER_FUNC1, and 10 sec press, power-off) worked.
However the charge passed 15% and nothing happened.
Battery ran out and got a forced-power-off (properly logged).
Reconnected the power and the RPi woke up at 60% as specified.

Looking at the condition for triggering low charge in _EvalCharge() a `>` has to be replaced by `>=`:
```
if level == 0 or ((level < th) and ((chargeLevel-level) > 0 and (chargeLevel-level) < 3)):
```
has to be
```
if level == 0 or ((level < th) and ((chargeLevel-level) >= 0 and (chargeLevel-level) < 3)):
```
After this change I got a crash because `lowChgEn` was not defined inside `_EvalCharge()`
`lowChgEn` has to be declared global at the top of `main()` and inside `_EvalCharge()`.

`lowBatVolEn` in `_EvalBatVoltage()` has exactly the same problem.

After these additional changes the low charge detection (finally) worked properly.

shawaj, 2018-01-14:
Thanks @tvoverbeek

DenisKhrutsky, 2018-01-15:
Changes applied in [Commit 93b7a63](https://github.com/PiSupply/PiJuice/commit/93b7a636fceaa6a2ea278510eca243692afa81c3)

closed 2018-01-15 by DenisKhrutsky