PiSupply/PiJuice #42: LEDs not behaving as expected

francesco-vannini, 2018-01-18:
Setting one of the RGB components in the GUI for D1 doesn't change it's colour. Furthermore after applying the new settings the green status light disappears. The blue charging light stays no matter what happens.
It all goes back to normal if resetting the board via "Reset to default configuration".

Also noticed that when messing with the D1 RGB settings in the GUI I get communication errors.
Suppose the update LED command to the STM32 interferes with the STM32 LED commanding.
Probably also what happens when the green disappears.
Something to fix in the firmware?

@mmilann this is related to what I was telling you about lately. What do you think?
@tvoverbeek could you post the errors you were talking about?

francesco-vannini, 2018-01-18:
@mmilann provided additional information for this @tvoverbeek Unlike what I thought the LED tab to be this is how it actually works and what the parameters are for.

For charge status function R,G,B parameters does not define color. Green parameter defines color component for charge level of green over 15% of charge, R parameter defines color component level of red below 50%. So between 15% and 50% you have mix of R and G. When battery is charging than blue blink is added to mix, level defined with B parameter, and when it is fully charged blue is always on.

I'll change the label as @shawaj was telling me that we might possibly want to rethink this page

francesco-vannini, 2018-01-18:
This is only happening when CHARGE_STATUS and ON_OFF_STATUS are selected, it USER_LED is chosen then the RGB fields work as expected on both D1 and D2. 
Using:
```python
pijuice.status.SetLedState('D1', [100,0,0])
```
will render D1 green and
```python
pijuice.status.SetLedState('D1', [255,0,0])
```
will make it brighter

tvoverbeek, 2018-01-18:
@mmilann @francesco-vannini Yes that seems to match what I am seeing.
However with the default settings 0f R60 G60 the Red LED completely dominates and you do not see the green LED between 15% and 50%.
With R at 60 and G at 70 I see the both the red and green LED between 50% and 15% and only the red LED below 15%.
Change the default setting to R 60 G 70 B 100 ?

Regarding the errors I am seeing. Occasionally when changing the RGB values in the LED dialog I get the "Apply LED Configuration" "Communication error" alert. Also the main status window shows 'Communication Error' for all the status values. Restarting the GUI clears this. Still suspect an issue in case the Tray app, the GUI and/or the service are trying to send I2C commands simultaneously.

tvoverbeek, 2018-01-18:
By the way I have never seen the always on blue (fully charged) even when the GUI and Tray say 100%.

shawaj, 2018-05-18:
@ChristopherRush when you have a moment do you think we could document all this and close this issue?

closed 2018-05-21 by ChristopherRush