PiSupply/PiJuice #44: Additional information for firmware

francesco-vannini, 2018-01-18:
The following items need to be documented.

Wakeup on charge updated to be activated only if power source is present. 
Futher this enables wakeup after plugged if this parameter is set to 0.

Button wakeup functions power off and power off can be now assigned to arbitrary button events. 
Removed constrain to be assigned only to long_press2 for power off, and single_press for power on.

Added reset function that can be assigned to some of buttons and button events.

Added no battery turn on configuration. 
Now it can be set whether or not user wants to turn on 5V rail as
soon as power input is connected and there is no battery.

Added configuration for 2 IO ports. 
They can be set to analog input, digital input, digital output and pwm output.

closed 2018-01-24 by francesco-vannini

shawaj, 2018-01-24:
Did these get fixed then?

francesco-vannini, 2018-01-25:
As far as I can tell they are in the documentation

shawaj, 2018-01-25:
Perfect!