PiSupply/PiJuice #45: Power on-off presets

francesco-vannini, 2018-01-18:
I think 10 seconds for LONG_PRESS1 associated to SYS_FUNC_HALT and 20 seconds for LONG_PRESS2 associated to HARD_FUNC_POWER_OFF are far too long.
I'd propose to preset LONG_PRESS1 to 5 seconds and LONG_PRESS2 to 10 seconds. This is mostly to do with the fact that once you power off the Pi the PiJuice won't automatically remove power to the Pi and people will have to use SW1 to do so.

shawaj, 2018-01-18:
It's configurable anyway no?

The default times were intentionally very long to avoid accidents really

tvoverbeek, 2018-01-18:
It is configurable, but as @francesco-vannini said there is no easy way to remove power to the Pi after a normal shutdown.

shawaj, 2018-01-18:
There is, PiJuice can do it automatically either by powering through PiJuice micro USB or by soldering the spring pin

shawaj, 2018-01-18:
I'm pretty sure Milan built this in. Although off the top of my head, I can't remember the exact method for doing it

tvoverbeek, 2018-01-18:
Yes there is. Configure LONG_PRESS1 to SYS_FUNC_HALT_POW_OFF instead of the default SYS_FUNC_HALT.It takes 60 seconds after the Pi shutdown before PiJuice removes the power to the Pi.
The 60 seconds is hard coded in pijuice_sys.py.
Which reminds me, all the SYS_FUNC_* functions need to be documented.
Also the switching off of external stuff connected to VSYS.

shawaj, 2018-01-18:
I guess that 60 seconds should ideally be configurable too, not immediately essential though as 60 seconds is probably OK for most scenarios

francesco-vannini, 2018-01-19:
My bad I didn't know you can use SYS_FUNC_HALT_POW_OFF.

shawaj, 2018-01-19:
All good, that's why we do this process of letting Ton rip it all apart :-P

shawaj, 2018-05-18:
@wikipedia555 can we change these hard coded times to be configurable?

tvoverbeek, 2018-05-18:
If you mean the time until power off, it is not hardcoded. It is the parameter when you select the SYS_FUNC_HALT_POW_OFF similar to the button press times.

shawaj, 2018-05-18:
@tvoverbeek you said above "The 60 seconds is hard coded in pijuice_sys.py." ?

Looks like there is some additional documentation to be done here as well

tvoverbeek, 2018-05-18:
You are right, I was mixing it up with the button press duration.
It is line 57 in pijuice_sys.py.
Here in context:
```
  elif func == 'SYS_FUNC_HALT_POW_OFF':
        pijuice.power.SetSystemPowerSwitch(0)
        pijuice.power.SetPowerOff(60)      <------
        _SystemHalt(event)
```
Could be made a variable which can be changed via the gui/cli and stored in the JSON file.