PiSupply/PiJuice #46: Stacking PiJuice

shawaj, 2018-01-18:
@kkrossa asked this...

I can not remember if the PiJuice can stack on an other PiJuice. If a person got three Pijuice would it be 12 hours working time on the RaspberryPi?

The answer is as follows:

It might be possible if disable 5V GPIO input so one board will not charge another, than only way to charge is through USB Micro input. But it is not efficient to do that way, much better to just use bigger battery. And I2C address should be changed but also rpi software does not support multiple pijuices(yet)

francesco-vannini, 2018-01-19:
Nice question but a bigger battery surely is the only reasonable way to go, right?

shawaj, 2018-01-19:
Define "reasonable"? I think it's reasonable for stacking, and we sell more PiJuice boards. So I don't mind.

But not urgent... Something for a rainy day

klint-k, 2018-01-20:
The best way is if you have the, 5V gpio and the power off, pins separated. If so then from the top down the battery's could discharge. charging would be a nightmare.

shawaj, 2018-01-21:
That's not really possible as the 5v rail is all connected and the boards are all the same...