PiSupply/PiJuice #49: Duplicate files and obsolete

francesco-vannini, 2018-01-19:
pijuicetest:
- https://github.com/PiSupply/PiJuice/blob/develop/Software/Source/pijuicetest.py
- https://github.com/PiSupply/PiJuice/blob/develop/Software/Source/src/pijuicetest.py

under Software/Source/doc there are two readme files which could be incorporated under the main READMEs instead. Are those for the Debian package structure?

Same for PiJuice\Software\man\pijuice.1

Same for PiJuice\Software\Source\README.md

Same for PiJuice\Software\Install\README.md

Could we not have these information incorporated in the main READMEs? If we need these to be there can we not still have all info in the main READMEs and only copy specific paragraphs of these into the files for Debian?
Having too many files like that around will be confusing and difficult to maintain.

shawaj, 2018-01-19:
Agreed. Could even reference, via a link, the online readme that has more details

francesco-vannini, 2018-01-19:
@DenisKhrutsky maybe you can help me out understanding what's to stay and how best to format these, no hurry.

francesco-vannini, 2018-01-25:
@DenisKhrutsky and @mmilann could you clarify why those files are there and if the content can be moved to the main READMEs? Do we need any of the above because of packaging requirements?

shawaj, 2018-01-25:
I think these were mainly added by me or Milan as guides for doing certain things to do with testing or packaging etc, and to not lose any information.

They're wasn't necessarily any reason for it other than that - so yes I'd agree we can probably move into the bigger readme for ease of maintenance

shawaj, 2018-01-25:
But yeah good to have input as to whether any of them are important...

francesco-vannini, 2018-01-25:
This is now sorted and all info have been transferred to the rightful pages

closed 2018-01-25 by francesco-vannini

shawaj, 2018-01-25:
Perfect!