PiSupply/PiJuice #50: Add a minimum time to the watchdog?

shawaj, 2018-01-19:
To avoid issues with it getting stuck in a reboot loop?

shawaj, 2018-01-19:
Or maybe just a warning if you put it below a certain value

mmilann, 2018-01-19:
That is accounted because watchdog is set in minutes,  1 - 65535 minutes
activates watchdog, 0 disables watchdog.

shawaj, 2018-01-19:
Ah ok so you can't go lower than 1 minute?

mmilann, 2018-01-19:
no, 1 minute is step.

shawaj, 2018-01-19:
Perfect, @francesco-vannini can we add this to documentation?

francesco-vannini, 2018-01-19:
Maybe best if one of you does it because I didn't quite understand this honestly. Otherwise if you can put it in a copy/paste format I can ... copy/paste it

shawaj, 2018-01-19:
The watchdog timer has a configurable time-out - the time after which it will reboot if it doesn't receive a heartbeat signal. The time step is in minutes so the minimum time-out period is one minute and the maximum is 65535 minutes. The number can be any whole number between one and 65535.

Does that make more sense @francesco-vannini ?

Also @mmilann, just to check, it is not possible to set this to 0 right? Minimum is definitely 1?

shawaj, 2018-01-19:
Reboot is probably incorrect. Maybe better to say power cycle. As basically it'll revive power to the Pi and then reapply it

shawaj, 2018-01-19:
*remove

mmilann, 2018-01-19:
It is possible to set 0, but it is decoded as watchdog disable.

shawaj, 2018-01-19:
Ok so there is no scenario whereby you can go less than 1 minute with
watchdog enabled?

francesco-vannini, 2018-01-22:
Added in 1a27ba99d1290b0a9b9d348cadc661f6c2d81a31

closed 2018-01-22 by francesco-vannini