PiSupply/PiJuice #52: Firmware updates via GUI

francesco-vannini, 2018-01-19:
From @mmilann:
"I do not recommend having browse button. Without authenticity check user
can load and file this will make non responding to any command and also
activating bootloader to re-update, so user must initiate bootloader
manually. Other reason is that new firmware will usually bring new features
that requires software update, so no sense to update only firmware without
software.

What can be good is to have all previous firmware in
same /usr/share/pijuice/data/firmware/ directory, and maybe having option
to revert to previous version."

shawaj, 2018-01-19:
Did you see my comment about this too?

francesco-vannini, 2018-01-19:
To which Aaron said: "Hmm. What can authenticity check look like? Alternatively we can have it so that instead of browse it can have "check for new firmware" and it looks for specific file name structure in firmware section on GitHub.

As you say, this removes possibility of people bricking their device.

And yes good idea about the previous versions thing!"

francesco-vannini, 2018-01-19:
Then Milan replied: "Simplest authenticity is to have checksum calculated over binary file and
than put calculated value inside file name.

Checking for updates when gui is launched, or periodically, for example
weekly and show it in tray status, but update whole software package, and
firmware inside."

francesco-vannini, 2018-01-19:
I'm probably the best secretary you'll ever have guys, I can copy/paste like none else :)

shawaj, 2018-01-19:
You wanted a career change :-P

Seriously though, maybe a combination of all three things - checksum, update checks on load and weekly, notification in tray, host new versions on GitHub and provide ability to roll back to old versions easily

tvoverbeek, 2018-01-20:
If you foresee more frequent updates to the firmware than to the software we should decouple the version numbering of the software and firmware.
As @mmilann already suggested (`update whole software package`)when a new firmware is available the debian package should be updated. Debian package versioning has a suffix which you increment when there is only a single change (i.e. new firmware). The main version of the package (now at 1.1) does not need to change.

francesco-vannini, 2018-01-22:
OK for this release I suggest we keep things as we have implemented in the dev branch because we are running out of time. We'll keep this open for the next release.

shawaj, 2018-01-22:
Ok sure

mmilann, 2018-01-22:
Add at least checking browsed file satisfy firmware file naming condition
to decrease possibility of random file, or completely remove browse button
if out of time and implement in next release.

shawaj, 2018-05-18:
@tvoverbeek I guess we can close this now? It seems like the firmware won't change more than the software, and as such we can just put any new firmware upgrades into the software releases for GUI based updates, and we can do "bleeding edge" firmware updates by the command line method?

tvoverbeek, 2018-05-18:
Yes, but the firmware is included in the base package (and should be), not in the gui package.

closed 2018-05-18 by tvoverbeek

shawaj, 2018-05-18:
agreed about it being included in the base version, and being able to upgrade from there

@tvoverbeek are there any changes you would suggest to the way the firmware updates are handled either by base / gui?

tvoverbeek, 2018-05-18:
Base is OK. GUI needs some kind of upload status screen ('Please wait, update in progress') which is foreseen in the code, but does not work.
Will work on this when back home.