PiSupply/PiJuice #53: LED ON_OFF_STATUS function not implemented in firmware

mmilann, 2018-01-21:
There is placeholder for ON_OFF_STATUS LED function in software but not implemented in firmware. This can confuse users and should be removed as option from configuration or to implement in firmware.

tvoverbeek, 2018-01-21:
I would suggest just delete the option from pijuice_gui.

shawaj, 2018-01-22:
Why was it added in the first place? Was it added by you @mmilann ?

What is it meant to show? Whether PiJuice is on or off?

mmilann, 2018-01-22:
I have added it as possibility for for future implementation in C enumeration of LED functions. In software it is added to list in order to easily translate to LED function code. One easy way to remove from software is to rename to NOT_USED, other to convert list to dictionary with only implemented names.

shawaj, 2018-01-22:
If it is useful to have, why don't we just add it to future roadmap for firmware?

DenisKhrutsky, 2018-01-22:
I've decided to keep the original list of functions and add the list of available options. If ON_OFF_STATUS has been set for some users, it will be set to NOT_USED. These sections has been marked in comments to easily remove these hacks when the option is going to be implemented.

shawaj, 2018-01-22:
Ok great. Well let's implement it in the next firmware release and then it's done

shawaj, 2018-05-18:
@mmilann was this implemented in the firmware in the end?

mmilann, 2018-05-18:
Not implemented, left for future based on feedback.