PiSupply/PiJuice #54: Use of IO pins on P3

tvoverbeek, 2018-01-22:
Nice that you can configure the IO pins in the GUI.
These are 2 unused GPIOs on teh STM32.
It is not documented how to get input and output from these STM32 GPIOS.

shawaj, 2018-01-22:
@mmilann any ideas about this?

mmilann, 2018-01-22:
There are api python functions implemented in pijuice.py for dynamic IO access. Here are usage examples:
print pijuice.status.GetIoAnalogInput(1) # Get analog input value from IO1 in milivolts,
print pijuice.status.GetIoDigitalInput(2) # Get digital input value from IO2
pijuice.status.SetIoDigitalOutput(1, 1) # Set digital output value 1 to IO1
pijuice.status.SetIoPWM(2, 35.6) # Set pwm duty circle to 35.6% for IO2
print pijuice.status.GetIoPWM(1) # Get current pwm duty circle at IO1

tvoverbeek, 2018-01-22:
Only added last week.
We still need to document the accessible user functions in pijuice.py (API document???)

shawaj, 2018-05-18:
@ChristopherRush could you take a look at this, when you have a moment?

ChristopherRush, 2019-10-15:
b55baa1569c55e519ca48b45e66627e45a4b0fff

closed 2019-10-15 by ChristopherRush