PiSupply/PiJuice #58: Building packages with pckg-pijuice.sh fails for pijuice-gui

tvoverbeek, 2018-01-24:
Trying to build the packages with pckg-pijuice.sh. This fails for pijuice-gui because setup complains pijuice.__version__ is not defined. This setup (copied from the main setup.py) runs in a lower directory where pijuice.py is not accessible.
Propose the following changes:
In pckg-pijuice.sh replace `PIJUICE_VERSION=$(python -c "import pijuice; print pijuice.__version__")`
with
`export PIJUICE_VERSION=$(python -c "import pijuice; print pijuice.__version__"`
(Here pijuice.__version__ is defined because pijuice.py is also in the Source directory).
In setup.py replace line 109 with:
`    version=os.environ.get('PIJUICE_VERSION'),`
WIth these changes, both packages build fine and version can still be defined in one place (pijuice.py).

DenisKhrutsky, 2018-01-24:
Build process was fine for me probably because of the compiled files. I'll test the build process and deal with this problem.

DenisKhrutsky, 2018-01-24:
Tried on clean OS and got this problem too. Thank you. Resolved in [#14d663](https://github.com/PiSupply/PiJuice/commit/14d663d905b7a3cee7f51041bce0786885ccde8c)

closed 2018-01-24 by DenisKhrutsky