PiSupply/PiJuice #62: Battery profiles in software section README?

shawaj, 2018-01-25:
@francesco-vannini is it intentional that at the bottom of the software readme there is a section about battery profiles? https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#battery-profiles

shawaj, 2018-01-25:
Maybe that data could be moved elsewhere and would be more at home?

Also - on this part about the battery config menu - https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#pijuice-hat-config-battery-menu

I think ideally we should provide more information about what all the settings are / where they come from...and crucially - which are optional and which are required?

francesco-vannini, 2018-01-25:
Yes that bit needs sorting, as for the GUI tab I'll add a link to https://github.com/PiSupply/PiJuice/tree/master/Hardware#battery-profiles

shawaj, 2018-01-25:
What do you mean about the GUI tab?

francesco-vannini, 2018-01-25:
The one you pointed out https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#pijuice-hat-config-battery-menu

shawaj, 2018-01-25:
I'm confused about what you're suggesting but ok :-)

francesco-vannini, 2018-01-25:
Sorted in 995216cccee26e668c34ac052f419e70541fa91c

closed 2018-01-25 by francesco-vannini