PiSupply/PiJuice #63: master software / firmware changelogs in readmes somehwere?

shawaj, 2018-01-25:
We have got a changelog in the debian packages, which is fine, but I am wondering if we should maintain a CHANGELOG.md file somewhere in the firmware and software folders just to keep a full doc, all in one place, with all the changes and updates in it.

francesco-vannini, 2018-01-25:
As I explained in #33 the intention was to keep the historical information of the changelog. I agree an overall changelog in https://github.com/PiSupply/PiJuice would make sense to keep and maintain

shawaj, 2018-01-25:
Maybe just a single document in the main folder with hardware, software and
firmware versions listed

Or individual documents in each of the sub folders

Which do you think is best?

tvoverbeek, 2018-01-25:
Prefer a single document. This makes it easier to check how firmware changes and software changes correlate.
Separate sections in a single document is probably better, because then the debian package changelogs can easily be produced from the software section.
It is always a pain when you have to maintain synchronised changelogs in different places.

francesco-vannini, 2018-01-25:
Sorted in 8643f590623061e7e799fd8db4943dcc77a725ca

closed 2018-01-25 by francesco-vannini