PiSupply/PiJuice #66: apt-get install not working

klint-k, 2018-01-26:
pi@raspberrypi2:~ $ sudo apt-get install -y --install-suggests pijuice-gui
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package pijuice-gui
pi@raspberrypi2:~ $ 
jan 25 20:20 MST

tvoverbeek, 2018-01-26:
Suppose you did not do an 'sudo apt-get update' first, since it was added to the repo recently.
It is there.
See the log of one my Pis which was not updated till today:
```
pi@lightning:~ $ apt-cache search pijuice
pijuice - Packaging a PiJuice Python script as DEB
pi@lightning:~ $ sudo apt-get update
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages [11.7 MB]  
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [127 kB]         
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [27.7 kB]                                                    
Get:6 http://mirrordirector.raspbian.org/raspbian stretch/contrib armhf Packages [56.8 kB]                                         
Get:7 http://mirrordirector.raspbian.org/raspbian stretch/non-free armhf Packages [95.2 kB]                                        
Fetched 12.0 MB in 32s (375 kB/s)                                                                                                  
Reading package lists... Done
pi@lightning:~ $ apt-cache search pijuice
pijuice-base - Packaging a PiJuice Python script as DEB
pijuice-gui - Packaging a PiJuice Python script as DEB
pi@lightning:~ $ 
```

klint-k, 2018-01-26:
sudo apt-get update
sudo apt-get -y upgrade
sudo apt-get -y dist-upgrade
Then the rest.

klint-k, 2018-01-26:
pi@raspberrypi2:~/script $ sudo apt-get update
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages          
Hit http://archive.raspberrypi.org jessie/main Sources                     
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages           
Hit http://archive.raspberrypi.org jessie/ui Sources                           
Hit http://archive.raspberrypi.org jessie/main armhf Packages                  
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages          
Hit http://archive.raspberrypi.org jessie/ui armhf Packages                    
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages               
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB               
Ign http://archive.raspberrypi.org jessie/main Translation-en                  
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB                 
Ign http://archive.raspberrypi.org jessie/ui Translation-en                    
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB        
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Reading package lists... Done
pi@raspberrypi2:~/script $ sudo apt-cache search pijuice
pijuice - Packaging a PiJuice Python script as DEB
pi@raspberrypi2:~/script $ apt-cache search pijuice
pijuice - Packaging a PiJuice Python script as DEB
pi@raspberrypi2:~/script $

tvoverbeek, 2018-01-27:
Currently pijuice-gui and pijuice-base are only available in Stretch.
Jessie has the original pijuice.
@shawaj can you try to get it also included in Jessie? SInce it is only python based there should not be any problems.

shawaj, 2018-01-27:
Yes will ask Serge how to do that...

shawaj, 2018-01-29:
ok all sorted now 

@klint-k could you try again on jessie?

shawaj, 2018-01-29:
or @tvoverbeek ?

tvoverbeek, 2018-01-29:
@shawaj, @klint-k,
Just tried on a jessie system. pijuice-gui and pijuice-base are not available yet.
Also checked directly on http://archive.raspberrypi.org/debian/dists/jessie/main/binary-armhf/
and the Packages file is not updated yet.
The packages are in the pool, but the pool is common for all available distributions.
Guess will have to wait till the archive update has been done,

shawaj, 2018-01-29:
yep, i think serge maybe has to do this...

tvoverbeek, 2018-01-30:
@shawaj @mmilann 
Could you check with Serge again about the inclusion of pijuice-base and pijuice-gui in Jessie?
I  noticed the packages in the pool were put there on Jan 24.
The Packages file for Stretch was updated yesterday (Jan 29)
The Packages file for Jessie was last updated on Nov 24, 2017 => No pijuice-base and pijuice-gui for Jessie.

shawaj, 2018-01-31:
Will do. But maybe best to do so once v1.2 is ready - or whatever the next
version is - as @francesco-vannini said this was due tomorrow?

shawaj, 2018-02-05:
v1.2 of GUI and Base are now in repo....well, I just uploaded them anyway.

@tvoverbeek @klint-k could you try again?

tvoverbeek, 2018-02-05:
@shawaj No sign of the 1.2 packages yet on archive.raspberrypi.org.
Do not know how long it takes from upload to general availability.
Did you specify both jessie and stretch?

shawaj, 2018-02-05:
@tvoverbeek yep, I did both. Hopefully should filter through soon. I guess it has to be approved by someone at RPF

shawaj, 2018-02-06:
Serge said this when i asked him about the packages going up to the repo:

> 5 minutes, but the part that automatically checks them and adds them to the repo isn't finished yet. Right now it will be up to 2 working days.

tvoverbeek, 2018-02-06:
Just checked. They are there now, both in Jessie and Stretch.
Will try the update later today.

tvoverbeek, 2018-02-06:
Tried install on Jessie (`sudo apt-get install pijuce-gui`) and upgrade on Stretch (`sudo apt-get upgrate`) after `sudo apt-get update` of course.
Worked as expected. No problems.
As far as I am concerned the issue can be closed.

shawaj, 2018-02-06:
Thanks @tvoverbeek

closed 2018-02-06 by shawaj

santisantis, 2021-12-03:
Hello,
has anyone experienced this problem on the following version of the Pi:
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

tvoverbeek, 2021-12-03:
Known problem. Pijuice packages are not included in the bullseye repository.
You can install pijuice-base from the buster repository or from github on bullseye.
pijuice-gui has a small problem (notification icon in the task bar for the battery level has a black background instead of white) which needs to be fixed before the pacakges can be included in the bullseye repository. The gui itself works fine under bullseye.

Smiles8212, 2022-01-25:
Hello tvoverbeek,
I have the same problem as santisantis. You say:

> You can install pijuice-base from the buster repository or from github on bullseye.

But I have abolutely no idea how to do that. Have you a step-by-step tuto or some extended explanations ?

Many thanks in advance!

tvoverbeek, 2022-01-25:
Ok here an attempt for a step by step guide to install pijuice-base on bullseye 32-bit:

- Download pijuice-base_1.7_all.deb from github: https://github.com/PiSupply/PiJuice/raw/master/Software/Install/pijuice-base_1.7_all.deb
- Install the python3-urwid dependency: `sudo apt-get install python3-urwid`
- Install pijuice-base manually from the directory where you downloaded it: `sudo dpkg -i ./pijuice-base_1.7_all.deb`

Smiles8212, 2022-01-25:
Hello Mr. van Overbeek
Thank you very much for your answer.
Now the pijuice-base run on the Raspi!
I had to install the i2c-tools too and also solve library dependences:
***@***.***:~ $ sudo dpkg -i /home/pi/pijuice-base_1.7_all.deb
Selecting previously unselected package pijuice-base.
(Reading database ... 41408 files and directories currently installed.)
Preparing to unpack .../pi/pijuice-base_1.7_all.deb ...
Unpacking pijuice-base (1.7) ...
dpkg: dependency problems prevent configuration of pijuice-base:
 pijuice-base depends on i2c-tools; however:
  Package i2c-tools is not installed.
 pijuice-base depends on python3-smbus; however:
  Package python3-smbus is not installed.
dpkg: error processing package pijuice-base (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 pijuice-base"
But after the installation of " i2c-tools" all is fine.
--
Meilleures salutations / Best regards / mit freundlichen Grüßen
Yann Monbaron
http://www.digi-train.ch/
----Message d'origine----

GerBreOwn, 2022-08-27:
It is now Aug 27, 2022. I am using Raspian Bullseye on a RPI Zero WH and the program I have that was running on the previous version of Raspian now no longer runs on Bullseye. I keep getting the error: " import pijuice ModuleNotFoundError: No module named 'pijuice". Any ideas on how to fix this problem?
Thanks

tvoverbeek, 2022-08-27:
Bullseye only has python3. If your program assumes/specifies python2.x it will not work.

GerBreOwn, 2022-08-28:
My version of Python is 3.10.5 as V2 has had EOL a long time ago.

tvoverbeek, 2022-08-28:
@GerBreOwn The current pijuice-base package assumes Debian Bullseye with python 3.9.x
It install pijuice.py in /usr/lib/python3.9/dist-packages and a symbolic link to /usr/lib/python3.9/dist-packages/pijuce.py in /usr/lib/python3/dist-packages so all python 3.x versions should be able to find it.
A previous python2.7 version will not be found when using python 3.x.
Please check your /usr/lib/python3.9/dist-packages and /usr/lib/python3/dist-packages.
Also check your python search path (sys.path from the sys package)

GerBreOwn, 2022-08-29:
Like I said before I am using Python3.10.5. Does this mean I have to go BACK to 3.9 to get pijuice to run?

Last night I removed pijuice 1.7 and 1.8. Tonight I reinstalled 1.8. I checked /usr/lib/python3.9/dist-packages and there is "pijuice.py" and "pijuice_base_1.8.egg-info" files there. In "/usr/lib/python3/dist-package" there is a file "pijuice.py" dated  27 Jan 2022 and it show "link" as file type but it does not show what it is linked to.
One problem I discovered was my PYTHONPATH was pointing to "/usr/lib/python3/dist-" so added "packages" to that and then it found pijuice, but then it said " No module named 'smbus'". When I tried to install python3-smbus it said "python3-smbus is already the newest version (4.2-1+b1)." I found it in "Requirement already satisfied: smbus2 in /home/gerald/.local/lib/python3.10/site-packages (0.4.2)"
Looks like there is a conflict between Python3.9.x and Python3.10.x. I have read somewhere that Python is NOT backward compatible.
Any suggestions other than going back to Python3.9 to solve these problems?
Thanks.

tvoverbeek, 2022-08-29:
No need to go back to 3.9.
To see where a file is linked to use the following command in a terminal (not GUI filemanager):
`ls -l /usr/lib/python3/dist-packages/pijuice.py`.
smbus2 is a smbus replacement in pure python, smbus uses C code interfacing directly to the kernel.
Either remove the smbus2 package and then install python3-smbus, or edit pijuice.py and replace line 9 with:
`from smbus2 import SMBus`
You obviously do no have a standard Raspberry Pi OS, but something else (Ubuntu?)

GerBreOwn, 2022-08-29:
**Thanks for the quick response**. **Adding "2" to smbus did the trick**.
I am using the standard Raspian Bullseye software but it is on a Raspberry Pi **Zero** WH computer. On my RPi4 I do have Ubuntu running.
The Zero is gathering the data from my weather station and then sending it via WIFI to the RPi4 where it will be generating the web pages.

tvoverbeek, 2022-08-29:
Glad you got it working. You probably have installed the smbus2 package as part of an other I2C device software.
The official Raspberry Pi OS repositories only have the smbus package.
Anyway good luck with your weather station project.