PiSupply/PiJuice #68: How to shut down/start on battery %

francesco-vannini, 2018-02-26:
So I've found pijuice-base in the Stretch repo, and tried to configure it, using the documentation. I came as far as having the pi shut itself down after booting it when the service was running, but not knowing why as the charge level was > 60%. You can find  the configuration below.
Basically what I'm trying to achieve is that the pi shuts down when the battery is low (say, 10%), and to boot up again when it's full enough to not shut down if power is lost immediately after we started booting and power is lost.

Some questions: (Ultimate question: How do I achieve my goal?)
1. What's the exact meaning of the SYS_FUNC_* flags? Does SYS_FUNC_POW_OFF also shut the pijuice down? Why is it mentioned twice in the documentation? This is also a request in #47  
2. I've configured the min_charge to 10, expecting it to trigger the "low_charge" system event when pijuice.status.GetChargeLevel() drops below 10. This was not the case. What am I doing wrong?
3. Similar to pijuice.status.GetChargeLevel(), is there also a way to obtain the voltage, so I can configure min_bat_voltage in a way that makes sense?
4. I found the unit of system_event.wakeup_on_charge.trigger_level to be in volts, ranging 0 to 10. Is this the battery voltage? How does it relate to min_bat_voltage?
5. Do I have to write the wake_up_on_charge stuff to the pijuice myself (using PiJuiceConfig?), or is that done when the pijuice service notices a change in the configuration file?

`/var/lib/pijuice/pijuice_config.JSON's` content:
```
{
  "system_events": {
    "low_battery_voltage": {
      "function": "SYS_FUNC_HALT_POW_OFF",
      "enabled": true
    }, 
    "low_charge": {
      "function": "SYS_FUNC_HALT_POW_OFF", 
      "enabled": true
    }, 
    "button_power_off": {
      "function": "SYS_FUNC_HALT_POW_OFF", 
      "enabled": true
    }, 
    "forced_power_off": {
      "function": "SYS_FUNC_HALT_POW_OFF", 
      "enabled": true
    }, 
    "no_power": {
      "function": "NO_FUNC", 
      "enabled": true
    }, 
    "forced_sys_power_off": {
      "function": "SYS_FUNC_HALT_POW_OFF", 
      "enabled": true
    }, 
    "watchdog_reset": {
      "function": "SYS_FUNC_REBOOT", 
      "enabled": true
    }
  }, 
  "user_functions": {
    "USER_FUNC8": "", 
    "USER_FUNC1": "", 
    "USER_FUNC2": "", 
    "USER_FUNC3": "", 
    "USER_FUNC4": "", 
    "USER_FUNC5": "", 
    "USER_FUNC6": "", 
    "USER_FUNC7": "",
    "USER_FUNC8": "",
    "USER_FUNC9": "",
    "USER_FUNC10": "",
    "USER_FUNC11": "",
    "USER_FUNC12": "",
    "USER_FUNC13": "",
    "USER_FUNC14": "",
    "USER_FUNC15": ""
  }, 
  "system_task": {
    "watchdog": {
      "enabled": true, 
      "period": "60"
    }, 
    "min_bat_voltage": {
      "threshold": "10", 
      "enabled": true
    }, 
    "min_charge": {
      "threshold": "10", 
      "enabled": true
    }, 
    "enabled": true, 
    "wakeup_on_charge": {
      "enabled": true, 
      "trigger_level": "5"
    }
  }
}
```

francesco-vannini, 2018-02-26:
@DenisKhrutsky had some comments:
1. What's the exact meaning of the SYS_FUNC_* flags?
SYS_FUNC_HALT: sets wakeup_on_charge (if it's enabled) before halting (sudo halt)
SYS_FUNC_HALT_POW_OFF: sends POWER_OFF_CMD and SYSTEM_POWER_SWITCH_CTRL_CMD
SYS_FUNC_SYS_OFF_HALT: sends POWER_OFF_CMD
2. No idea for now. Code for this case seems fine.
3. pijuice.status.GetBatteryVoltage()
5. Simple way to update configuration is to restart the pijuice service after your changes in config file.
Another way is to send SIGHUP signal (which is used by GUI application to notify service about changes) to pijuice_sys.py process after your changes: kill -HUP $(pgrep pijuice_sys.py)
I am working on updating pijuice code to python 3 right now. Needed a break from the CLI app.

francesco-vannini, 2018-02-27:
@ChristopherRush came back with
Here are my JOSN settings, all done in the GUI. Note: Still fully testing this out.

Under System Task:
Wakeup on charge - 50% (When PiJuice battery is above this the system will boot)
Minimum charge - 10% (When the PiJuice drops below this it will trigger System Event > Low charge
Under System Events:
Low Charge - SYS_FUNC_HALT_LOW_POW_OFF (Shuts Pi down safely)

```
{

  "system_events": {

    "low_charge": {

      "function": "SYS_FUNC_HALT_POW_OFF",

      "enabled": true

    }

  },

  "user_functions": {

    "USER_FUNC8": "",

    "USER_FUNC1": "",

    "USER_FUNC2": "",

    "USER_FUNC3": "",

    "USER_FUNC4": "",

    "USER_FUNC5": "",

    "USER_FUNC6": "",

    "USER_FUNC7": ""

  },

  "system_task": {

    "watchdog": {

      "period": ""

    },

    "min_bat_voltage": {

      "threshold": ""

    },

    "min_charge": {

      "threshold": "10",

      "enabled": true

    },

    "enabled": true,

    "wakeup_on_charge": {

      "enabled": true,

      "trigger_level": "50"

    }

  }

}
```

francesco-vannini, 2018-02-27:
@mmilann 
Config file from Chris looks fine for automatic shutdown/wakeup on charge levels.

Here is screenshot of config that trigger shutdown when charge drops below 3% 
and wakes-up when charge goes above 90%.

Charge level precision drops as load current rises and that case advice is to set enough
low/high trigger levels for stability reasons.
[image]

akeeman, 2018-03-13:
Hi @francesco-vannini,

I've tried the configuration mentioned [here](https://github.com/PiSupply/PiJuice/issues/68#issuecomment-368779470). It indeed works a lot better. However, there are a few problems left:
* The measured charge level in my setup is quite unreliable. The measured charge level is greatly dependent on whether Pi runs on battery power or not. It can switch from <10% to near 100% in minutes after turning on external power, to drop again to <10% in less then a minute after switching external power off. Or in a more extreme case, after the Pi has switched off because of a <10% charge level, it will boot again spontaneously because a charge level of >50% has been measured some how, without turning on the external power.
* The boot process can be interrupted for some reason, which causes the Pi to not automatically boot itself again. This happens sometimes after the Pi shuts down due to a charge level of <10%, then I turn on the external power, then the Pi starts booting after a little while. But during that boot process it sometimes shuts off instantly (so I guess the PiJuice cuts the power). After this the Pi won't boot again without pressing the power button.

Any suggestions on how to fix these?

shawaj, 2018-03-17:
@mmilann any ideas?

mmilann, 2018-03-19:
Charge level measurement precision is decreased by increased load current
due to internal resistance of protection circuit inside BP7X battery. It is
like
logarithmic dependence, so there can be more than 50% charge when it is
actually around 10% if load current is enough high.

For better stability of automatic shutdown feature triggered by "Low charge"
event with BP7X battery use lower settings of "Minimum charge", as indicator
when battery is close to empty. Make experiments to find satisfying value
in
your application or alternatively use "Minimum charge"/"Low battery voltage"
feature.

"Wakeup on charge" feature is intended to be used in field applications
where
system is powered from harvesting source connected to pijuice USB micro
input to automatically wake-up system when power source appears (sunlight),
and battery charges above defined level. In this case when Rpi is off, load
current is low and charge level measurement precision is good.

shawaj, 2018-03-28:
@akeeman did you see the response from @ChristopherRush on this?

shawaj, 2018-03-28:
Just seen that there was an email I missed.

Copying here for reference:

Hi Maarten,

I've been running a number of tests with both the Raspberry Pi 2 and the new Raspberry Pi 3 B+ board with the PiJuice. I have connected in both tests a USB HDD and a USB Wi-Fi adapter to simulate the amount of current that may be used. I configured the PiJuice to shutdown when below 2% and to boot up when it was 99% charged. Both scenarios proved similar results as the battery lasted about 1hr 20mins with the Pi 3 B+ and 1hr 45 mins with the Raspberry Pi 2 before it shutdown, which is as expected with a 1820 mA battery and a 1.1-1.4A load (load fluctuated). Both times the board did a clean shutdown and also a good boot when fully charged (99%).

Having run these tests I am fully confident that the PiJuice battery levels are correct and are producing the correct readings as and when the board is in use based on the peripherals that are connected. Based on the current load and battery size the discharge times are accurate and as expected. The key factor in these tests were the configuration settings of the PiJuice, the Raspberry Pi shuts down when below 2% and then bots up when at 99%. When the PI boots up again the values drop to about 70% and stabilise for the reminder discharge duration based upon the load.

Please let me know if what I have stated makes sense and if you are happy with those results. If there is anything in particular you want me to look at then also let me know.

Chris

shawaj, 2018-03-28:
And a couple of follow up questions:

What about the life expectancy of the batteries?

When you reboot at 99% is that still viable after, lets say a year or so, when maybe only 80% of the battery remains?

Or does it then take 99% of the remaining 80% of the battery’s capacity.

shawaj, 2018-05-18:
@ChristopherRush this is now fixed isn't it? Do you think it is sufficiently documented here?

ChristopherRush, 2018-05-21:
@shawaj All the sys functions are now documented, as well as further explanation of GUI menu's.

smartexpt, 2018-07-24:
@shawaj and @ChristopherRush 

I'm having similar trouble with the auto shutdown and power on feature by setting the `min_charge` to 10% and `wakeup_on_charge` to 99%.

All changes were made via CLI/JSON file, which by the way looks like this:
```
{
        "system_events": {
                "low_charge": {
                        "function": "NO_FUNC",
                        "enabled": true
                }
        },
        "system_task": {
                "watchdog": {
                        "enabled": true,
                        "period": "5"
                },
                "min_charge": {
                        "threshold": "10",
                        "enabled": true
                },
                "wakeup_on_charge": {
                        "enabled": true,
                        "trigger_level": "90"
                },
                "enabled": true
        }
}
```
From what I've read, this issue should be solved setting very low values for `min_charge` and very high values to `wakeup_on_charge`. Am I correct?

For these features to work properly, should the power come from the PiJuice HAT?

Thanks in advance.

tvoverbeek, 2018-07-25:
@smartexpt 

>From what I've read, this issue should be solved setting very low values for min_charge and very high values to wakeup_on_charge. Am I correct?

wakeup_on_charge does not have to be **very** high, a value of 60% or 70% will also work.
Note that wakeup_on_charge will only work if the pijuice shuts down the RPi.

>For these features to work properly, should the power come from the PiJuice HAT?

Yes, for obvious reasons. When you power via the RPi, the RPi continues to supply 5V to the PiJuice via the GPIO and hence the PiJuice will continue keeping the battery charged.

>After update the JSON file manually, should I force the update somehow as previously said by @DenisKhrutsky , through kill -HUP $(pgrep pijuice_sys.py)?

Yes, otherwise the system task will not pick up the changed settings. Altrenatively you can reboot.

closed 2019-10-15 by ChristopherRush