PiSupply/PiJuice #70: Separate packages needed for Python 2 and Python 3

tvoverbeek, 2018-03-03:
We can support Python 2 and Python 3 with the same source files, but Raspbian/Debian requires separate packages for Python 2 and Python 3.
Normally the convention is to use python-<package name> for Python 2 and python3-<package name> for Python 3.
Now we have 'pijuice-base' and 'pijuice-gui' which is not compatible with this convention.
Note that stdeb can produce both Python 2 and 3 packages. Have not tried it yet.

Just some food for thought for the next version

shawaj, 2018-03-03:
> Normally the convention is to use python- for Python 2 and python3- for

Is that true even in the .deb packages?

I've never seen .deb packages with python- or python3- in front of them?

tvoverbeek, 2018-03-04:
There are plenty.
Here the results of `apt-cache search` for python2 and python3 packages in the 2 attached files.
[python3-packages.txt](https://github.com/PiSupply/PiJuice/files/1778358/python3-packages.txt)
[python2-packages.txt](https://github.com/PiSupply/PiJuice/files/1778359/python2-packages.txt)

Note the python- and python3- prefixes are only in the deb names.
Import statements still use the 'normal' names without any prefix.
This is needed because Python 2 looks for its packages in `/usr/lib/python2.7/dist-packages/`
and Python 3 in `/usr/lib/python3/dist-packages`
Default is still Python2 for Raspbian, which is why the current pijuice packages are only available for Python2.
Next release of Raspbian will probably default to Python3. So we should add support for Python3 asap.

shawaj, 2018-03-04:
Hmm. I mean I haven't been through them all but they seem like they're add
ons to extend functionality of things you can do with python.

Ours is a standalone interface for PiJuice and has no other use than for
that. And just happens to be written in Python.

Can we not just build a single Deb package that is agnostic to which
version of python you're using and just defaults to whichever is the
default on that device?

tvoverbeek, 2018-03-04:
> Can we not just build a single Deb package that is agnostic to which
version of python you're using and just defaults to whichever is the
default on that device?

Short answer: No.

You can have a single source producing both package versions provided the code runs under both Python2 and Python3.

This is what Pimoroni does for their hardware:
```
python2-packages.txt:python-blinkt - Python library for the Pimoroni Blinkt!
python2-packages.txt:python-buttonshim - Python library for the Pimoroni Button SHIM
python2-packages.txt:python-drumhat - Python library for the Pimoroni Drum HAT.
python2-packages.txt:python-fourletterphat - Python library for the Pimoroni Four Letter pHAT
python2-packages.txt:python-inkyphat - python library for the Pimoroni Inky pHAT
python2-packages.txt:python-pantilthat - Python library for the Pimoroni Pan-Tilt HAT
python2-packages.txt:python-phatbeat - Python library for the Pimoroni pHAT BEAT.
python2-packages.txt:python-scrollphathd - Python library for the Pimoroni Scroll pHAT HD.
python2-packages.txt:python-unicornhathd - Python library for the Pimoroni Unicorn HAT HD.
python3-packages.txt:python3-blinkt - Python library for the Pimoroni Blinkt!
python3-packages.txt:python3-buttonshim - Python library for the Pimoroni Button SHIM
python3-packages.txt:python3-drumhat - Python library for the Pimoroni Drum HAT.
python3-packages.txt:python3-fourletterphat - python library for the Pimoroni Four Letter pHAT
python3-packages.txt:python3-inkyphat - python library for the Pimoroni Inky pHAT
python3-packages.txt:python3-pantilthat - Python library for the Pimoroni Pan-Tilt HAT
python3-packages.txt:python3-phatbeat - Python library for the Pimoroni pHAT BEAT.
python3-packages.txt:python3-scrollphathd - Python library for the Pimoroni Scroll pHAT HD.
python3-packages.txt:python3-unicornhathd - Python library for the Pimoroni Unicorn HAT HD.
```

shawaj, 2018-03-05:
But they are calling those "python libraries" - this isn't a python library
it's standalone software that just happens to be written in Python. Surely
we can just package it with either or both.

I appreciate that might not be the "rules" of python based software but it
would be a much cleaner way to do this for end users.

tvoverbeek, 2018-03-05:
Approaching this from a more pragmatic point of view:
The only 'problematic' file is pijuice.py, the low level interface which all the others import.
pijuice.py is installed in the python version specific dist-packages directory.
All the other 'applications' (pijuice_sys.py, pijuice_gui.py, pijuice_cli.py and pijuice_tray.py) are
installed in /usr/bin.
If a user has set its default python to python3 (meaning /usr/bin/env python will start python3)
the applications will start but will fail because pijuice.py cannot be found/imported.

Practical solution might be in the postinstall script of pijuice-base to copy pijuice.py also to /usr/lib/python3/dist-packages.
@DenisKhrutsky what do you think?

DenisKhrutsky, 2018-03-06:
I agree with @shawaj that we have a standalone application and separating into two can cause unwanted problems. But we cannot simply copy the library into python3 lib folder because some packages can be missing. I had an idea about keeping the pijuice library and pijuice_sys separately. But I don't know yet how to properly handle the installation in that case. Can we specify python-pijuice OR python3-pijuice somehow or we the packages can only depend on both?

shawaj, 2018-03-07:
Maybe there is a way to detect the default python version? And then change
behaviour based on that?

klint-k, 2018-03-07:
how about finding out which python they are using to install?

tvoverbeek, 2018-04-08:
Even simpler:
```
import sys
version=sys.version_info()
```
pijuice.py, pijuice_tray.py, pijuice_gui.py and pijuice_sys.py already run under both
python 2 and python 3 without changes.

Still looking for a good way to install using raspbian packages.

shawaj, 2020-02-22:
@tvoverbeek I think we can probably close this now we've moved over to python 3. Would you agree?

tvoverbeek, 2020-02-23:
Yes, closing this issue

closed 2020-02-23 by tvoverbeek