PiSupply/PiJuice #72: "Charging enabled" is unchecked by default?

shawaj, 2018-03-15:
A customer has just reported that the "Charging Enabled" is unchecked by default and as such the battery does not charge until this setting is turned on.

@DenisKhrutsky @mmilann @tvoverbeek @ChristopherRush @francesco-vannini have any of you seen this as well?

ChristopherRush, 2018-03-15:
I have just done a fresh install and can confirm that it is enabled by default. @shawaj @tvoverbeek @francesco-vannini

tvoverbeek, 2018-03-15:
I have seen it (Charging disabled) on my Kickstarter PiJuice.
The 'Charging Enabled' is stored in the PiJuice processor.
I saw it after upgrading the software from 1.0 to 1.1.
I guess (if I remember correctly) the charging enabled was introduced in firmware 1.1 (@mmilann correct?). If so it depends on the state of the status-bit in the shipped Kickstarter PiJuices with firmware 1.0 since the bit was not used in 1.0.

mmilann, 2018-03-15:
Charging disable is introduced in firmware V1.0_2017_10_23. I checked this
version and latest from
V1.1_2018_01_15, both are enabled with fresh, non-configured unit.

For earlier firmware without charge disable feature, it is always enabled
no matter what is configuration.

tvoverbeek, 2018-03-16:
I only know I ended up with my Kickstarter PiJuice after updating to 1.1 with charging disabled.
I was using the 1.3.1-1 version of the software with settings from my sample PiJuice.
Maybe some unlucky combination of software and firmware versions and stored settings?

mmilann, 2018-03-17:
Will test those cases after firmware update. There is configuration space
in mcu EEPROM
that is preserved, content is not cleared during update, and might cause
false config
in case address structure was different in earlier release.

shawaj, 2018-05-18:
@tvoverbeek @ChristopherRush I think this is fixed in the latest firmware / software combination?

Are you happy if we close this?

ChristopherRush, 2018-05-18:
Yes, can confirm it is fixed. Close

closed 2018-05-18 by tvoverbeek

McJohnPi, 2019-06-02:
A couple of weeks ago we installed version 1.4 of the PiJuice software. 
Last week we needed the PiJuice but it was totally empty and won't charge. 
We take a look at all the settings but we can't find the problem., all the power input was the same as before.
And now we saw suddenly the checkbox "Charging Enabled" was  standard unchecked in the 1.4 update.... 😬