PiSupply/PiJuice #73: Cannot shutdown, how to debug?

tinue, 2018-03-16:
I am using a standard PiJuice with the official battery, no pogo-pin, stacked onto a Pi 3. It is powered through the PiJuice Micro-USB. The software was installed, the firmware upgraded to 1.1 through the GUI. The GUI shows the charge state of the battery, so I'd say the communication with the PiJuice is working.
SW1 also works from a hardware point of view, because I could initially switch on the Raspberry Pi by clicking the switch.

What I cannot do is power off the Pi. It doesn't work by pressing SW1 for 10 Seconds, it doesn't work after I lowered the value to 5 Seconds, it doesn't work by using SW2 or SW3 after configuring both do do "SYS_FUNC_HALT_POWER_OFF" on "SINGLE PRESS". Whatever I do, the Pi doesn't budge and remains on. The PiJuice also doesn't cut power after 60 seconds.

How can I debug this? When I check the sources on GitHub, logging seems to be disabled.

I am using the latest Raspbian, fully updated (kernel 4.9.80, March 9).

tvoverbeek, 2018-03-16:
Smells like the pijuice service is not running.
All the functions you describe as not working need the pijuice service (pijuice_sys.py) running.
The hard power off and initial switch on only need the PIC on the PiJuice board.

What does 'systemctl status pijuice.service' report?
Here is my output:
```
pi@rpi-stretch-full:~ $ systemctl status pijuice.service
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-03-15 21:50:06 CET; 20h ago
 Main PID: 399 (pijuice_sys.py)
   CGroup: /system.slice/pijuice.service
           └─399 /usr/bin/python /usr/bin/pijuice_sys.py

Mar 15 21:50:06 rpi-stretch-full systemd[1]: Started PiJuice status service.
pi@rpi-stretch-full:~ $ 
```

tinue, 2018-03-16:
I see the same output. Also, /var/run/pijuice.pid does exist and correlates with a running process:
```
me@pijuicepi:~$ ps -ef | grep 427
root       427     1  0 16:55 ?        00:00:00 /usr/bin/python /usr/bin/pijuice_sys.py
```

tvoverbeek, 2018-03-16:
You can try to run the service manually.
First stop the pijuice service: `systemctl stop pijuice.service`.
Then in a terminal: `sudo python /usr/bin/pijuice_sys.py`.
This could produce some output which might help.

Also are you running the updated pijuice-base and pijuice-gui packages?
My output from `dpkg -l 'pijuice*'` is:
```
pi@rpi-stretch-full:~ $ dpkg -l 'pijuice*'
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                        Version            Architecture       Description
+++-===========================-==================-==================-===========================================================
un  pijuice                     <none>             <none>             (no description available)
ii  pijuice-base                1.3.1-1            all                Packaging a PiJuice Python script as DEB
ii  pijuice-gui                 1.3.1-1            all                Packaging a PiJuice Python script as DEB
pi@rpi-stretch-full:~ $ 
```
1.3.1-1 is the latest.

tinue, 2018-03-16:
The versions match. I did put in a few print statements, and ran the script as you suggested.
It appears that this line always evaluates to 'false':
`if configData.get('system_task', {}).get('enabled'):`

So the script just idles in the 'dopoll' loop, doing nothing.

tvoverbeek, 2018-03-16:
In the pijuice gui (Preferences -> PiJuice Settings on the Raspbian desktop) is the System Task enabled?
See attached screenshot.
[image]

It is the first item on the System Task tab.

tinue, 2018-03-17:
It's not enabled; Enabling this fixes the problem. Thanks!

I'm not quite sure why this happened. I did install the base package first, because I didn't think that I need the GUI. After the buttons did not work, I installed the GUI package.

The installer seems to not touch the config file if one is already there.
So my best guess is that the base package by default does not enable the system task, and so the buttons do not work by default with the base package.

tvoverbeek, 2018-03-17:
@tinue Glad it works now.
Indeed, when installing the base package a default config file (/var/lib/pijuice/pijuice_config.JSON) is created with the system task disabled.
Probably time to change this to enabled.

tvoverbeek, 2018-03-17:
See commit https://github.com/PiSupply/PiJuice/commit/8eebe90dea41cd8d708588546d137d52e5534a03
Closing issue

closed 2018-03-17 by tvoverbeek