PiSupply/PiJuice #75: Charging never stops

tinue, 2018-03-17:
My PiJuice has the BP7X battery installed. According to the battery profile, the "Regulation voltage" is 4.18 V.

When the micro USB cable is plugged into the PiJuice, the charge level never reaches this value. It may go as high as 4.165 V, then drop again to 4.15 V. The blue charge LED always blinks.

Is the battery profile wrong? Or is the measurement off by a few Millivolts?

I could make my own battery profile I guess to solve this.

tvoverbeek, 2018-03-17:
When I have the RPi powered it never reaches full charge (blue/green blinking).
With the RPi powered off completely (All Pi LEDs off) and the PiJuice charging I eventully
get a solid blue = fully charged.
I am not a battery expert, but I guess with the RPi powered (even with only the power LED on and
shut down) it stays in trickle charge.

mmilann, 2018-03-17:
When charged from pijuice USB micro input, there is load current consumed
by rpi drawn from battery rail that is usually higher than termination
current
(50mA default), so in this case in voltage regulation phase charging just
does
not terminates, even if it is fully charged,. When load current drops (rpi
shutdown)
or when powered from rpi power input, charger can detect when charge
current
drops below termination threshold and it signals with steady blue.

tinue, 2018-03-18:
Thank you both for the observation / explanation. Indeed, when the Pi is powered off the charging does stop.

I still don't understand the charge LED: It seems to be a bit useless. The battery voltage is known (it is visible on the GUI). Also, the battery "Regulation Voltage" is known from the battery profile.

I would turn off the charge LED when the battery voltage reaches the regulation voltage (or maybe 100-200 mV lower). This way, the user knows when the battery is "full" without having to open the GUI.

This would work regardless of how (and if) the board is powered.

What do you think?

shawaj, 2018-03-18:
@tinue - this could be a more intuitive way for it to work yes. Whilst it's not strictly speaking "fully charged" it is as close to being fully charged as it will ever be. 

An alternative could be a solid blue when fully charged with pi off and another solid colour when fully charged with pi on. Just a thought

mmilann, 2018-03-19:
There are tree charging phases. "Pre-charge phase" to recover if battery is
totally
discharged. Then comes "Constant current phase", where charge current is
regulated
to defined level until voltage over battery reaches regulation voltage.
Third phase is
"Voltage regulation phase" where voltage over battery is regulated to
defined regulation
voltage settings. Battery can be considered full only when in third phase
charge current
drops below termination current. Small difference between regulation
voltage settings
and what is measured over battery( few tens of milivolts) when it is
practically full is just
voltage drop over parasitic resistances when there is load current drawn
plus measurement
tolerances.

For UPS applications where most of time Rpi is powered by stable mains
source it is
recommended to use Rpi USB micro input. In this case you will have more
efficient powering
and charging termination will be clean. Also one tip in this case, setting
"Voltage regulation" to
lower value may increase battery life, but reduce total charge in battery
what is less important
for UPS.

tinue, 2018-03-20:
Thanks for the hint! I'm not yet clear whether I will use it outside, solar powered, or as a UPS backed mini server. If the latter I'll solder the pogo pin then.

For my education: If you don't know the current going to the battery (only the overall current going to the battery rail), how do you run the "constant current" phase of the charge cycle? The battery profile shows an exact value for the charge current, do you just add an average Pi power consumption amperage to this value?

This would mean a battery in a Raspberry Pi Zero gets charged in a shorter time compared to a battery in a Raspberry Pi 3 Model B+, right?

mmilann, 2018-03-20:
There is switch between battery and load that is "off" during constant
current phase
so charging current to battery is regulated closely to what is programmed.
Switch is
"on" in voltage regulation phase so load current is shared with charge
current.

If load power demand during "Constant current" phase is greater than input
power
minus charge power, than charge current is automatically reduced or event
supplied
back from battery to satisfy load. So battery is optimally charged as
programmed only
if there is enough input power, if not charging is prolonged.

tinue, 2018-03-21:
Thanks again for the explanation! Since I have no more questions I close the issue.

closed 2018-03-21 by tinue

tvoverbeek, 2018-03-21:
@mmilann's explanations should somehow be included in the hardware README