PiSupply/PiJuice #78: Wrong day displayed on date/time

Jonovich, 2018-03-19:
For some reason mine has the correct date, but is displaying the wrong day. If I correct it via the GUI by pressing "Set system time" when I then press 'Apply' I get an error:-

Failed to communicate with PiJuice service 
See system logs and Systemctl status pijuice.service for details.

Which report

● pijuice.service - PiJuice status service 
Loaded: loaded (/lib/systemd/system/pijuice.service; disabled; vendor preset: 
Active: inactive (dead)

Also, the error message is in a dialogue box titled "PuJuice" and not "PiJuice".

Jono

tvoverbeek, 2018-03-20:
Displaying the wrong weekday on the 'Wakeup Alarm' tab is a bug in pijuice_gui.
The Python calendar module numbers weekdays from 0 to 6, where 0=Monday and 6=Sunday.
The RTC in the PIC processor numbers them obviously (I have to check) from 1 to 7 where 1=Sunday and 7=Saturday.
The code makes the wrong conversion.

When you do 'Set system time' the time (in UTC) on the Pi is copied to the RTC in the PIC.

The pijuice service not running is an other problem. Cannot guess from here why it crashed.

Thanks for catching the spelling error on the error box title.

tvoverbeek, 2018-03-20:
The pijuice_gui code turns out to be correct.
The processor on the PiJuice board has a RTC where weekdays are coded 1(=Monday) to 7(=Sunday) -> 1 higher as the coding in the Python calendar weekday.
Assuming the time on your Pi is correct (including timezone and daylight saving/standard time) you need to copy the Pi time once to the PiJuice processor RTC using the 'Set system time' button on the 'Wakeup Alarm' tab.
Note this works directly, no need to click the 'Apply' button.

tvoverbeek, 2018-03-20:
Following Chapter in this story.

Noticed on my PiJuice clicking the 'Set system time' put the weekday on the 'Wakeup Alarm' tab to 'Tue' (It is Tuesday today).
Tuns out Raspbian detects an RTC at I2C address 0x68 and loads the rtc_ds1307 kernel module.
Since my PiJuice RPi is connected to the internet Raspbian regularly updates the RTC with the system time (systemd-timesyncd).
To see this you can also do this manually with 'sudo hwclock -w' which updates the RTC from the system time in UTC the same way as systemd-timesyncd.

This update uses the convention 1=Sun .. 7=Sat for the RTC weekday register.
The Python datetime module uses the convention 0=Mon .. 6=Sun
To stay compatible with the kernel module we need to use the following conversions:
```
RTC -> Python:
wday_python = (wday_rtc + 5) % 7

Python -> RTC
wday_rtc = (wday_puthon + 1) % 7 + 1
```
(% is the modulo operator)
@DenisKhrutsky What do you think?

mmilann, 2018-03-20:
PiJuice has device tree overlay in HAT EEPROM that automatically loads DS1307 Linux driver.
So it is possible to use system features for setting RTC and Alarm as @tvoverbeek explained.

Here are some tips to setup Alarm:
#rtc alarm status:
cat /proc/driver/rtc
cat /sys/class/rtc/rtc0/wakealarm

#set alarm in 1 minute:
sudo sh -c "echo `date '+%s' -d '+ 1 minutes'` > /sys/class/rtc/rtc0/wakealarm"

#clear alarm
sudo sh -c "echo 0 > /sys/class/rtc/rtc0/wakealarm"

Setting Alarm from GUI is alternative way, but more featured.

mmilann, 2018-03-20:
Actually it is DS1339 RTC emulation, version of DS1307 with alarm.

shawaj, 2018-03-20:
Just FYI @tvoverbeek I think "puthon" is a typo?

tvoverbeek, 2018-03-21:
@shawaj Yes, now corrected in the comment.

shawaj, 2018-03-21:
FYI @topshed is exepriencing the same date issue.

commit 7f482443 (2018-04-14): Fixes to pijuice, gui and tray. Major overhaul of pijuice_cli

pijuice.py

Change duty circle -> duty cycle
Convert button event parameter from string to integer (Fixes #xx)

pijuice_gui.py
Wakeupalarm: correct day of week conversion between RTC and Raspbian
Change 'Set system time' to 'Set RTC time' since the system time is
copied
to the PiJuice RTC.

pijuice_tray.py
Correct integer divisions to Python3 convention (//). Avoids crashing
when power connected to Pi and when on Battery (Fixes #78)

pijuice_cli.py
Fixed size to 64x20 so runs nicely on standard 80x24 terminal
Use inversion for sele [... cut]
commit 7fff184f (2018-12-18): Gui and tray updates (#94)

* Fixes to pijuice, gui and tray. Major overhaul of pijuice_cli

pijuice.py

Change duty circle -> duty cycle
Convert button event parameter from string to integer (Fixes #xx)

pijuice_gui.py
Wakeupalarm: correct day of week conversion between RTC and Raspbian
Change 'Set system time' to 'Set RTC time' since the system time is
copied
to the PiJuice RTC.

pijuice_tray.py
Correct integer divisions to Python3 convention (//). Avoids crashing
when power connected to Pi and when on Battery (Fixes #78)

pijuice_cli.py
Fixed size to 64x20 so runs nicely on standard 80x24 t [... cut]