PiSupply/PiJuice #79: Repeated wakeup not working

topshed, 2018-03-21:
I've been testing the PiJuice and I'm having some problems understanding how the Wakeup capability works.

I can make the Pi wake up once, but not repeatedly.  Perhaps I'm misunderstanding the GUI. On the Wakeup Alarm tab I check the boxes for "Every day", "Every hour" and then select the "Minutes period" radio button. I fill in the value 10 in the box beneath and leave the "Second" value at zero. Then I check the "Wakeup enabled" box and then hit Apply. Now my thinking is that the Pi should then power on every 10 minutes - is this correct? If I shut down my Pi, it does not power on again.

If I check the radio button for "Minute" instead,  then the Pi will power on at that value minutes passed the hour. But only once. So if the Pi powers on at 11:10 and I shut it down again, it does not power on again at 12:10

I'm not sure what the "Set Alarm" button does on this tab, but clicking it doesn't seem to make any difference to the outcome.  

What settings should I use if I want my Pi to power on every hour, at 10 minutes past the hour (assuming I then shut it down again using cron)?

shawaj, 2018-03-21:
@mmilann @DenisKhrutsky FYI

mmilann, 2018-03-22:
For wake-up every hour every day at 10 minutes past the hour, use setup from screenshot.
[image]
After configuring click on "Set Alarm" button, apply does not relate to this tab.

Minutes period should wake_up periodically at sub-hour level, this needs further inspection at firmware level to discover if there is bug.

tvoverbeek, 2018-03-22:
Here my experiences.
Everything reported below is with power connected to the PiJuice, since I do not have the pogo-pin mounted.

WIth the settings above (10 minutes past the hour) no boot when systemd-timesyncd.service is active.
Disabling systemd-timesyncd.service causes a reboot 10 past the hour, but only once.
After the system comes back up you have the correct the system time manually (since systemd-timesyncd is not active) with e.g. 'sudo hwclock -s'
Also after the system comes back up the check-box for 'Wakeup enabled' is not selected anymore.
If this check-box is off, no reboot.
Seems after a reboot some rearming of the alarm/boot mechanism has to occur.

If changing to minutes-period (e.g. every 5 minutes) no reboots occur.
It does not matter if 'Wakeup enabled' is selected or not.
However if you leave the system running the alarms do show up in the status row after 'Last:' every 5 minutes.
This works the same independent of systemd-timesyncd running or not.
After a manual reboot the alarms are still reported every 5 minutes on the status line.
So the periodic alarms do occur, but have no effect for booting the RPi.

@mmilann, @DenisKhrutsky Seems there is some work to do in the cooperation between the firmware and the software.

topshed, 2018-03-23:
Thanks @tvoverbeek - that is is exactly my experience too. I'd also add if that if I check the "minutes period" radio button, I don't ever get a wakeup restart. I have tried these setting both using the pogo pin and without it. It is not clear from the documentation under exactly which circumstance the wakeup facility will work.  I think the docs would benefit from a detailed example of how to get wakeup (say once every hour) working, including all the hardware config (e.g. pogo pin or not) required.

tvoverbeek, 2018-03-23:
First the firmware/software interaction has to be corrected before we can give a good example in the docs.

mmilann, 2018-03-28:
Here is new firmware that fixes non reactive alarm:
https://github.com/PiSupply/PiJuice/tree/master/Firmware
PiJuice-V1.2_2018_03_28.elf.binary
It can be tested by downloadin and runnig:
pijuiceboot 14 ./PiJuice-V1.2_2018_03_28.elf.binary

tvoverbeek, 2018-03-28:
@mmilann 
Did some tests using the V1.2_2018_03_28 firmware.
Some progress, but we are not there yet.
Tests were performed with power supplied via the PiJuice USB, no pogo-pin mounted.
The wakeups only work when the power to the Pi is removed.
With the Pi only shutdown (red power led still on) it does not work.
Specifying a minute-interval now wakes up the Pi.
The time based wakeup still seems to work only the first time.
Noticed that after reboot the wakeup checkbox is off.
You have to enable it again.
Also after some wakeups LONGPRESS1 on SW1 does not work any more.
Have to reapply the SW1 configuration.
Will do some more detailed testing and report later.

On a side-note firmware update in pijuice_cli does not work (too many arguments specified for update_firmware)

tvoverbeek, 2018-03-28:
@mmilann The V1.2_2018_03_28 firmware still reports its version as V1.1 (very confusing).
The gui does a local update of the version string, but restarting it will show firmware version again as 1.1.
Loading the 1.2 firmware externally with pijuiceboot and then starting the gui, will show version 1.1 again for the firmware.
Could you please replace the current 1.2 version with one which reports the correct version nr.?
Thanks in advance.

hp1337, 2018-06-12:
Has there been a resolution to this issue?

tvoverbeek, 2018-06-12:
Yes.
When you leave the ID-EEPROM (the one specifying the HAT configuration to Rapbian) at its default address 0x50, Raspbian at boot when processing the device-tree recognizes a ds1339 i2c real time clock (which is simulated in the PiJuice firmware). Raspbian at RTC clock initialization will then reset the bit in the firmware which specifies the wakeup-enable capability.
In this case you have to re-enable the wakeup-enable with a simple script (started from i.e. /etc/rc.local or as a cron job at boot).
For a full working example see https://github.com/PiSupply/PiJuice/issues/91#issuecomment-381324352
The essential part is the following:
```
import pijuice
pj = pijuice.PiJuice(1, 0x14)

# Enable wakeup, otherwise power to the RPi will not be
# applied when the RTC alarm goes off
pj.rtcAlarm.SetWakeupEnabled(True)
```
Alternatively you can change the ID-EEPROM address to 0x52, so the HAT configuration is not read by Raspbian when processing the device-tree.
Then the wakeup-enabled bit will remain set over reboots.

hp1337, 2018-06-15:
Great thank you, I got it working with your example in #91.

For the benefit of others: If one adds their own script to run at restart through the root crontab, the Raspbian RTC clock initialisation occurs around the same time that the user's script. As a result it becomes indeterminate if the SetWakeupEnabled(True) will get overwritten or not. To fix this issue, I added a delay to the beginning of my script. This guarantees that the RTC clock initialisation occurs before and then my script will be guaranteed to appropriately SetWakeupEnabeled(True).

Thanks!

tvoverbeek, 2018-06-15:
@hp1337 Yes, you need a delay since the cron job is started very early in the boot sequence, before the i2c devices are created.
In my example in #91 I use the following code to wait for the i2c-device:
```
# This script is started at reboot by cron.
# Since the start is very early in the boot sequence we wait for the i2c-1 device
while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)
```
Will update the example and README on github accordingly.

hp1337, 2018-06-15:
@tvoverbeek I found that waiting for the i2c device was not enough. Sometimes the RTC initialization would occur after my script already executed. A longer delay was required. If there was a way to wait for the RTC initialization to complete that would be ideal.

tvoverbeek, 2018-06-15:
@hp1337 Thanks for the feedback.
Will try to find a way to wait for RTC init (waiting for /dev/rtc ?)

SkySlab, 2018-07-07:
I use a headless setup, Rpi model 2 B+ on Stretch and had to set the eeprom address to 52 using the cli interface (general -> eeprom address) to get the wakeup command to set on reboot.  Did not matter how much delay I used in the script or in cron, I even tried rc.local and building a 15 second delay into that.

tvoverbeek, 2018-07-07:
@SkySlab As explained above, when you leave the eeprom address at 50, the RPi reads the eeprom at boot and sees a DS1339 real-time clock. The initialization of the RTC clears the wakeup enable bit.
If in addition your Pi is connected to the internet, the timesyncd daemon regularly updates the RTC which also clears the wakeup-enable bit.
By changing the eeprom address to 52 no RTC is recognized at boot and hence the wakeup-enable bit remains set.

closed 2019-10-23 by ChristopherRush