PiSupply/PiJuice #81: document error states?

shawaj, 2018-03-27:
A kickstarter backer posted this:

TobbeJ, 2018-03-27:
yes documenting what counts as bad and good would be nice, lower and upper limits.
i suspect that in my case probably the usb battery pack gives a little more voltage than expected by the pijuice and then it is flagged as bad.

my experience with the usb battery in question is actually quite good with other devices since it allows 5v 2amp output to charge for example a phone or smilar at good speed.
so i would not count the usb battery as bad.

shawaj, 2018-03-27:
@TobbeJ I guess like all error states, it is only in reference to our use case the sense of "bad" rather than it actually being a crappy PSU. Really it kind of means "out of spec"

Hopefully @mmilann or maybe @DenisKhrutsky can reply about what this actually means

And any other error states we should be aware of :-)

TobbeJ, 2018-03-27:
one more thing to document.
can you please write something about "Charging enabled" on this page: https://github.com/PiSupply/PiJuice/blob/master/Software/README.md
and possibly also explain to me why it defaults to being off.
feels a bit counter intuitive.

since i have no gui i'm using pijuice_cli.py
"reset to default" turns off charging enabled.

shawaj, 2018-03-27:
@tvoverbeek we fixed this charging enabled thing didn't we m

shawaj, 2018-03-27:
Oh no, not yet, it's in issue #72

jf-ch, 2018-03-27:
I have a question about the statuses: When I connect the power to the RPi, I'm not able to communicate with the Pijuice. When the power is connected to the pijuice, everything is fine. 
Then I move the power cable from the pijuice to the RPi (without rebooting), and I have access to the pijuice but the status is:
[image]

Why this behavior? what means a negative amp?

shawaj, 2018-03-27:
@jf-ch what makes you think that it is not communicating with the Pi? It has a battery voltage and % so is fine. Also please post in a new issue any separate things please :-)

tvoverbeek, 2018-03-27:
@TobbeJ Regarding the 'Charging disabled' issue.
The default for the 1.1 firmware is charging enabled.
Are you running the 1.0 firmware (the one which is installed on the PiJuice when shipped)?
Check the firmware version. With pijuice_cli it shoudl say Version 1.1.
Make sure you run the update to 1.1.
After to the update you may still have 'Charging disabled' (happened to me).
If so change it to 'enabled'.

jf-ch, 2018-03-27:
@shawaj OK I will do it.
(because pijuice_cli.py return errors and exit)

TobbeJ, 2018-03-28:
@tvoverbeek first thing i did was to update to firmware 1.1
when i noticed charging was disabled i enabled it and fount charging to work, then i tested to reset all settings to see if it was just a fw update issue but after resetting all settings via pijuice_cli charging was again disabled.

shawaj, 2018-03-28:
@jf-ch ok thanks. Look forward to seeing that. Do you mean the firmware update doesn't work because I think we fixed that in #77 ?

tvoverbeek, 2018-03-29:
firmware update with pijuice_cli does not work. This part was obviously never tested.
Only way without the gui now is to do it manually: 'pijuiceboot 14 <path-to-firmwarefile>'

jf-ch, 2018-03-29:
I updated pijuice_base to 1.3.2-1 thru apt last week and was able to update successfully my pijuice to 1.1

tvoverbeek, 2018-03-29:
Yes, works in 1.3.2-1 (Had not updated pijuice_cli yet).
Still some problems because pijuiceboot output destroys the urwid screen.

shawaj, 2018-03-29:
@tvoverbeek so the pijuice_cli does work or doesn't? I am a little confused as you seemed to say it didnt above but then say it does? Has it been fixed in 1.3.2-1?

tvoverbeek, 2018-03-29:
My 'does not work' comment was with manually updated versions of pijuice-*.py directly from github.
But I had not picked  up the change in pijuice_cli.
With that change the update works, although it needs improvements (output from pijuiceboot should be made invisible as in pijuice_gui)
Anyway under stretch you still get 1.3.1-1 with the non-working firmware-update in pijuice-cli.

TobbeJ, 2018-03-31:
the error states for starters need to be better documented.
as an example i connected the pijuice solar panel to micro usb on pijuice board and it counts as bad, connecting my inline usb based volt & amp meter shows 4.9v and 0mA

i would expect voltage to drop once load is applied if there is not much sun but why not use the power that is available?

also it was sunny when i did my test.

TobbeJ, 2018-03-31:
> @TobbeJ the solar panel won't put out anything unless it can achieve over 250mA. So that's likely the cause of what you're seeing

well... i don't believe that's possible, how can the panel "know" it can supply 250mA load if you have next  to no load connected to it?
what is more likely is that once load is applied by pijuice starting to charge then voltage collapses and pijuice marks usb input as bad then it stops, panel recovers and some time later pijuice tries again and process repeats possibly due to not much sun or other factors.

also i observed different charging behavior when pijuice & pi is on versus when everything is off.
when things is off then it occasionally starts to charge ( blue & green flashing led)
when pi & pijuice is on then it never goes into charging and usb input is marked as bad.

shawaj, 2018-04-01:
@TobbeJ the solar panel has internal circuitry, that's how. It's contained in the connector and behind the panels. They're defined to also charge smartphones which is why they are like this. If you want to trickle charge from any mA you can connect a standard solar panel without this restriction if you would like

shawaj, 2018-04-01:
*designed

mmilann, 2018-04-02:
@TobbeJ  try setting higher DPM from configuration GUI. @shawaj there is description in internal documentation about input DPM but is missing online.

shawaj, 2018-04-02:
@mmilann can you email me where this is?

TobbeJ, 2018-04-03:
> @TobbeJ try setting higher DPM from configuration GUI. @shawaj there is description in internal documentation about input DPM but is missing online.

changing "USB micro IN DPM" from default 4.2 to for example 4.52 appears to improve the charging situation from my usb battery at least for a little while anyway.
but there is something odd with pijuice_cli because when i change it and use apply it says settings was applied but it is not updated in the ui, same if i exit and go back it still lists 4.20v and i think batter still counts as bad after a while.

TobbeJ, 2018-04-03:
@shawaj ok, so let me get this straight.
you wrote:
> @TobbeJ the solar panel won't put out anything unless it can achieve over 250mA. So that's likely the cause of what you're seeing

Then i assume this means if i can measure ANY voltage (most likely it will be 5v due to the voltage regulator in the usb plug on panel) on the solar panel then i MUST be able to use 250mA from panel, right?
That's what you are saying.

This makes no sense to me, because i can measure a voltage with no load on the panel but the moment even a little load is applied the voltage collapses because panel can't keep up, for example under low light conditions.

What i suspect you meant was that panel needs enough sun light so it can output around 250mA for it to be stable enough to charge things.

shawaj, 2019-02-14:
@TobbeJ no i believe there is a circuit in the solar panel which does not activate the output until a certain current threshold is met. Are you seeing lower current values come through then?