PiSupply/PiJuice #82: battery draining when pijuice off?

shawaj, 2018-03-27:
lso another question, with pijuice connected to Pi zero (the one without wifi) and you have an external usb wifi adapter connected and you turn everything off, then i can still see a green led on the wifi adapter even after pi is off and pijuice flashing slowly. 
was green flashing but now i assume the usb wifi have drained the battery a bit because it is now flashing red instead (about 8-10 hours after power off).

@mmilann any ideas about this?

TobbeJ, 2018-03-27:
maybe i also can ask something related to this issue.
how do you know if you have held down the power button on pijuice for 10 seconds to trigger a soft power off?
it is not very practical to use a clock or time it or something for a simple soft power off

what i did was to hold down the power button until the "color flash" (think it was on the other led) that appeared around the 10 second mark then it looked like the pi zero shut down, pi power led was off and then later i noticed the led from usb wifi adapter was still on (it is a little hard to see)
so maybe this was a case of pi shutting down but not quite right so power was left on from pi juice.

so maybe i should ask then, if you do a soft power off, is there a timeout from pijuice when it will cut power anyway, for example if something went wrong during shutdown and it got stuck.

tvoverbeek, 2018-03-27:
@TobbeJ The default for a long press on SW1 (10 sec) is to execute SYS_FUNC_HALT.
This only shuts down the Pi, but does **not** remove the power to the Pi.
The Pi-Zero does not have a power led, only an activity led.
Hence your WiFi adapter will still draw power.
Assuming you are using pijuice-cli.py go to Buttons, select SW1 then function LONG_PRESS1.
Change function SYS_FUNC_HALT to SYS_FUNC_SYS_OFF_HALT.
If you want you can change the parameter 10000 to 5000 (5 sec press instead of 10 sec).
When LONG_PRESS1 is triggered on SW1 the Pi will shut down and the power to the Pi will be removed a minute later.

tvoverbeek, 2018-03-27:
Note: When you change the button configuration it will not be picked up immediately.
This is because the pijuice service only reads the button configuration when started.
Either reboot or restart the service: `sudo systemctl restart pijuice.service`.

TobbeJ, 2018-03-28:
@tvoverbeek thanks i will try that

one followup question, can you document SetPowerOff() function on your software page?
my guess is that it tells pijuice to cut power after specified number of seconds and that's what makes SYS_FUNC_HALT_POW_OFF work

TobbeJ, 2018-03-31:
changing to SYS_FUNC_HALT_POW_OFF works but when i try to change time from 10000 to 5000 pijuice_cli crashes with:

    Traceback (most recent call last):
      File "/usr/bin/pijuice_cli.py", line 1229, in <module>
        urwid.MainLoop(top, palette=[('reversed', 'standout', '')]).run()
      File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 274, in run
        self.screen.run_wrapper(self._run)
      File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 268, in run_wrapper
        return fn()
      File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 339, in _run
        self.event_loop.run()
      File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 669, in run
        self._loop()
      File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 706, in _loop
        self._watch_files[fd]()
      File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 390, in _update
        self.process_input(keys)
      File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 490, in process_input
        k = self._topmost_widget.keypress(self.screen_size, k)
      File "/usr/lib/python2.7/dist-packages/urwid/container.py", line 588, in keypress
        *self.calculate_padding_filler(size, True)), key)
      File "/usr/lib/python2.7/dist-packages/urwid/decoration.py", line 621, in keypress
        return self._original_widget.keypress(maxvals, key)
      File "/usr/lib/python2.7/dist-packages/urwid/decoration.py", line 621, in keypress
        return self._original_widget.keypress(maxvals, key)
      File "/usr/lib/python2.7/dist-packages/urwid/listbox.py", line 987, in keypress
        key = focus_widget.keypress((maxcol,),key)
      File "/usr/lib/python2.7/dist-packages/urwid/wimp.py", line 535, in keypress
        self._emit('click')
      File "/usr/lib/python2.7/dist-packages/urwid/widget.py", line 464, in _emit
        signals.emit_signal(self, name, self, *args)
      File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 264, in emit
        result |= self._call_callback(callback, user_arg, user_args, args)
      File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 294, in _call_callback
        return bool(callback(*args_to_pass))
      File "/usr/bin/pijuice_cli.py", line 590, in _apply_settings
        error_msg = pijuice.config.SetButtonConfiguration(button, self.current_config[button]).get('error', 'NO_ERROR')
      File "/usr/lib/python2.7/dist-packages/pijuice.py", line 1294, in SetButtonConfiguration
        data[i*2+1] = (config[self.buttonEvents[i]]['parameter'] // 100) & 0xff
    TypeError: unsupported operand type(s) for //: 'str' and 'int'

tvoverbeek, 2018-03-31:
Confirmed. This is a bug in pijuice.py
As a quick fix change line 1294 in '/usr/lib/python2.7/dist-packages/pijuice.py' from
`            data[i*2+1] = (config[self.buttonEvents[i]]['parameter'] // 100) & 0xff`
to
`            data[i*2+1] = (int(config[self.buttonEvents[i]]['parameter']) // 100) & 0xff`
@mmilann take note.

ChristopherRush, 2019-09-15:
Fixed here - https://github.com/PiSupply/PiJuice/commit/7fff184fd2da93693c5526f462b24f4152fb3a7e

closed 2019-09-15 by ChristopherRush