PiSupply/PiJuice #83: How to power 7" Touch Screen

johnshearing, 2018-03-28:
I have a Smarti Pi set up with the official 7" touch screen. It comes with a "**Y**" cable adapter that splits power from the AC power supply and delivers it to both the USB connector on the PiJuice and also to the USB connector on the 7" touch screen.  

If I pull the USB connector out of the PiJuice and leave the USB connector going to the touch screen inserted then the raspberry pi will still run fine on the battery while the touch screen is powered by the AC power supply. 

But to simulate a real power out situation I disconnect the USB connector on the AC power supply from the **Y** cable adapter. This leaves the USB on the PiJuice and the USB on the touch screen connected via the **Y** cable adapter.  This crashes the raspberry pi. 

I am guessing that your intention is for the PiJuice to power both the raspberry pi and the touch screen from the battery when the power from the AC power supply is lost. So it seems reasonable to assume that I do not have it wired correctly. If that is the case, how should I deliver power to the touch screen from the PiJuice?

On the other hand perhaps you never intended for the PiJuice to power anything but the raspberry pi and you expect me to use a different uninterruptible power supply to power the touch screen. If that is the case then I need to know that this is the case.

Please advise,
Thanks, John

tvoverbeek, 2018-03-28:
From the README on the hardware page:
__________________________________________________________
**Max. current that the PiJuice will be able to supply**

The maximum current at 5V via the GPIO is 2.5A and via VSYS is 2.1A. This is also dependent on battery capacity. For BP7X battery for example we have measured around 1.1A at 5V GPIO and around 1.6A at VSYS output. To achieve the maximum of 2.5A you will need to use a battery of at least 3500mAh
__________________________________________________________
Powering both the screen and the RPi simultaneously will probably draw more than 1.1A.
Hence the undervoltage detection on the Pi will kick in and shut it down (not gracefully).
You will have to wait till a battery with larger capacity than the BP7X is available for the PiJuice.

johnshearing, 2018-03-28:
Hello @tvoverbeek,
Thanks for the quick response.
Looks like I will need to set up something else for the touch screen. 
That will be fine. 
I am just delighted to have smart power management for the pi

closed 2018-03-28 by johnshearing

shawaj, 2018-03-28:
@francescovannini did test this successfully with the touch screen if I remember correctly. I think just connecting the power to the touch screen to the Pi GPIO pins

johnshearing, 2018-03-29:
Thanks @francescovannini
I will give that a try and report back

bklevence, 2019-11-16:
Is GPIO a confirmed method?