PiSupply/PiJuice #84: Using a Battery Pack 3 x 2200mAh Lithium Ion 18650-sized cells in parallel

johnshearing, 2018-03-28:
I would like to use the battery described below:
3 x 2200mAh Lithium Ion 18650-sized cells in parallel.

It can be seen at the following link:
https://www.adafruit.com/product/353

Is it possible to use this battery pack?
If so, how should I wire it to the PiJuice?

Thanks, John

tvoverbeek, 2018-03-28:
Short answer: No, only single cell batteries are supported.
Longer answer from the README on the hardware page:
___________________________________________________________
**Which batteries can be used with PiJuice**
You can use any single cell lipo with PiJuice as far as you configure the board and the software correctly. You cannot use cells/batteries in parallel or series. As mentioned above we recommend that you use a battery with an NTC temperature sensor on board as well as on board protection circuitry (overcharge / overcurrent etc). Please note that you use your own battery at your own cost and risk and we do not officially support using it in this configuration as there are far too many variables - so you are on your own if there are any issues or if you break your Pi or PiJuice or damage anything else.

We intend to create add on boards as accessories in the future which will allow use with other battery technologies but we do not have anything like this right now.
__________________________________________________________

johnshearing, 2018-03-28:
Hi @tvoverbeek 
Ok, that's fine. Now that I know I am only powering the pi with the PiJuice battery I can set up another battery to power other things in the project on a separate circuit.
Thanks so much for the help.

closed 2018-03-28 by johnshearing