PiSupply/PiJuice #85: commit comment, readme updates

mwrich4, 2018-03-28:
Each commit or updated should have brief notes so that changes can be independently verified and tested.  Helper will do this when good information is provided by all.  Lacking this, it will be an abandoned mess.

shawaj, 2018-03-28:
We already do this, but thanks anyway

closed 2018-03-28 by shawaj

mwrich4, 2018-03-28:
odd that the firmware change from today has no comment - huh

shawaj, 2018-03-28:
@mwrich4 - not sure where you are looking but I can see comments...

```
Adding updated firmware V1.2

Alarm fix. 
Initialization fix after firmware update and configuration reset.
```

shawaj, 2018-03-28:
You can see it here - https://github.com/PiSupply/PiJuice/commit/d2787a952b585126251bcd487c9051a997490ca3

Also discussed in #79 further

mwrich4, 2018-03-29:
'Alarm fix.'  
Where I work that wouldn't fly .. need to say what was fixed so the verifier/tester can make sure that it works as intended.   
(this is guesswork) Something like:  Alarm fix: USB voltage alarm now correctly responds to voltage too high / voltage too low.  Before voltage too high triggered a fault when above 5.1 volts.

shawaj, 2018-03-29:
@mwrich4 thanks for your input. we will take it into consideration for future.

mwrich4, 2018-03-29:
I made edit to last comment.  All I can think of.

tvoverbeek, 2018-03-29:
Note, this firmware version should not have been posted on github (my private opinion), since it is still under test to fix the problems described in issue #79 
The 1.2 version posted still has problems and introduced a new one (incorrectly reporting its version).