PiSupply/PiJuice #86: E: Unable to correct problems, you have held broken packages.

shawaj, 2018-03-28:
Reported by @CleoQC on Twitter:

Raspbian Stretch latest

pi@pi3bplus:~ $ sudo apt-get install pijuice-gui
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 pijuice-gui : Depends: pijuice-base (>= 1.3) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

shawaj, 2018-03-29:
I thought that too but @CleoQC said that it was a fresh SD card so not sure.

ChristopherRush, 2018-03-29:
I have a customer through FreshDesk who also cannot install. Not sure if its related or not, I have requested further info.

shawaj, 2018-03-29:
Maybe the latest update to the Deb package caused an issue?

tvoverbeek, 2018-03-29:
The latest update 1.3.2-1 is still not available via apt-get update/upgrade in stretch.
However 1.3.2-1 is available for jessie.
Are the people having problems still on jessie?

CleoQc, 2018-03-29:
Thanks for opening the issue for me @shawaj !

It was a fresh SD card, burned just for this. The Stretch image from March 13, running on a Pi3B+
However, I did change a few things before attempting to install pi-juice
First is the ubiquitous sudo apt-get update/upgrade . Even though the image is from March 13, there was a lot already to upgrade

Then in raspi-config, I changed a few things although I would be surprised if any of these would have an impact. I'm listing them simply for disclosure. 
Camera, SSH and VNC are enabled.  SPI, I2C I didn't touch.
Localisation is set to North America (locale, timezone, keyboard, and wifi country)
Those changes should be fairly safe  - which is why I claimed the SD card was a fresh install. 

PiJuice was the first package I attempted to install.

tvoverbeek, 2018-03-29:
Could reproduce the problem.
It occurs if you try to 'sudo apt-get install pijuice-gui' immediately after the first big update/upgrade/reboot.
Note this upgrades the kernel to 4.14.30 from March 26 2018.
For some reason it cannot find the python-urwid package which is needed by pijuice-base (for the pijuice_cli utility).
Running 'sudo apt-get update' again makes python-urwid available.
Check with 'apt-cache search python-urwid'
With python-urwid available you can run 'sudo apt-get install pijuice-gui' which will install
pijuce-gui, pijuice-base and python-urwid.

I guess we/you got caught by an update of the main archive which was not finished yet.

CleoQc, 2018-03-30:
Thanks. That would be an easy enough solution. I'm off to try it!

closed 2018-03-30 by tvoverbeek

cmsimike, 2018-05-02:
Thanks! Worked for me too. Same issue but solved with an update(??)