PiSupply/PiJuice #88: Wish: a real command line tool (NOT text based ui)

TobbeJ, 2018-03-31:
Would be nice to have a real command line version of what currently exists as pijuice_cli.py
so you for example could do something like:
pijuice_something --status
and other things, maybe command line option to output the different readings or possibly make configuration changes.

reason for this is that it allows easy scripting of things.
for example logging of battery level or power draw via a shell script and cron.

shawaj, 2018-05-18:
@wikipedia555 do you think we can do this?

tvoverbeek, 2018-05-18:
As @hp1337 wrote, I believe we should document the low level functions of pijuice.py, then people can write their own scripts without the need for pijuice_gui and pijuice_cli.
WIth a pijuice_cli fully equivalent with pijuice_gui, we do not need a pure commandline tool.

shawaj, 2018-05-18:
@tvoverbeek ok great, i agree this sounds like the best option.

guess this is a job for @wikipedia555 and @ChristopherRush then

patgg, 2019-07-21:
it would be more easy to have a command line to get status etc from the pijuice.
i saw something about that here : fovea1959/pijuice_util
but i tryied and it fails.
what should be good is some basics command line to get the status of the battery, to if is in charge or not.

tvoverbeek, 2019-07-23:
@patgg The fovea1959/pijuice-util does not work because nowadays the pijuice.py module is python3 only.
I edited a little bit and this runs now under python3. File attached
[pijuice_util.txt](https://github.com/PiSupply/PiJuice/files/3422342/pijuice_util.txt)
Rename to pijuice_util.py
Note values for voltage and current etc. are raw and not scaled.

patgg, 2019-07-24:
Hey perfect !
it is what i needed

> Note values for voltage and current etc. are raw and not scaled.

i am working on these values with shell script.
i ll put it later when finished.

tilghman, 2020-07-06:
[pijuice_util.txt](https://github.com/PiSupply/PiJuice/files/4880643/pijuice_util.txt)

I've added on dump and load options.

shawaj, 2020-10-24:
Closing this as we have #496 and #518

closed 2020-10-24 by shawaj