PiSupply/PiJuice #91: Shutting down from Python

odeckmyn, 2018-04-10:
Guys,

I think i've read the whole Internet looking for this information. Please excuse my question if there is an obvious answer somewhere already.

I'm trying to do the following with my Pi Zero W attached to my fresh PiJuice : 

- boot
- run a Python script that will : 
  - involve some tasks for say 2 minutes
  - program PiJuice to wakeup in 10 minutes (current time+10 minutes ; precision doesn't need to be great)
  - shutdown everything so that (almost) NO BATTERY is used during the next 10 minutes. A simple "shutdown -h now" won't fit (PiJuice is still sending current to the Pi, and the attached USB are still draining the battery)
- have PiJuice awake the Pi and run the script again, etc.

Technically I'm still facing 2 issues : 
- having a Python script initiate a full shutdown with a real power-off
- having a Python script program the next wakeup

Thanx for your help !

jf-ch, 2018-04-12:
I can tell you what I've done, which can may be cover your first part (in my case, I shutdown and power off the Rpi when a power outage is detected and power it on when the power come back):
```
pijuice.power.SetWakeUpOnCharge(0)
pijuice.power.SetPowerOff(40)
subprocess.call(["sudo", "halt"])
```

tvoverbeek, 2018-04-14:
@odeckmyn Here a test program which does what you want.

The test does a dummy job for 1 minute and then shuts down for 5 minutes.
The test is started by cron upon reboot.
Here is the crontab entry:
```
@reboot /usr/bin/python /home/pi/test.py
```
Since cron starts the job very early in the boot sequence, some of the drivers (e.g. i2c which we need for PiJuice) might not be loaded yet.

While the Pi is off the system clock is halted. Because the job duration is short (here 1 minute) there is not enough time for timesyncd to update the system time from the network. We have to rely on the RTC and copy the RTC time to the system clock using the hwclock utility.

Here is the complete test program in Python:
```
from __future__ import print_function

import time
import pijuice
import subprocess
import datetime
import os
import sys 

DELTA_MIN=5

# Rely on RTC to keep the time
subprocess.call(["sudo", "hwclock", "--hctosys"])

# Record start time
txt = datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S') + ' -- Started\n'
with open('/home/pi/test.log','a') as f:
    f.write(txt)

# This script is started at reboot by cron.
# Since the start is very early in the boot sequence we wait for the i2c-1 device
while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

try:
    pj = pijuice.PiJuice(1, 0x14)
except:
    print("Cannot create pijuice object")
    sys.exit()

# Do the work
for i in range(60):
   print('*', end='')
   sys.stdout.flush()
   time.sleep(1)
print()

# Set RTC alarm 5 minutes from now
# RTC is kept in UTC
a={}
a['year'] = 'EVERY_YEAR'
a['month'] = 'EVERY_MONTH'
a['day'] = 'EVERY_DAY'
a['hour'] = 'EVERY_HOUR'
t = datetime.datetime.utcnow()
a['minute'] = (t.minute + DELTA_MIN) % 60
a['second'] = 0
status = pj.rtcAlarm.SetAlarm(a)
if status['error'] != 'NO_ERROR':
    print('Cannot set alarm\n')
    sys.exit()
else:
    print('Alarm set for ' + str(pj.rtcAlarm.GetAlarm()))

# Enable wakeup, otherwise power to the RPi will not be
# applied when the RTC alarm goes off
pj.rtcAlarm.SetWakeupEnabled(True)
time.sleep(0.4)

# PiJuice shuts down power to Rpi after 20 sec from now
# This leaves sufficient time to execute the shutdown sequence
pj.power.SetPowerOff(20)
subprocess.call(["sudo", "poweroff"])
```
Here the resulting test.log file:
```
2018-04-13 21:40:34 -- Started
2018-04-13 21:46:12 -- Started
2018-04-13 21:52:13 -- Started
2018-04-13 21:58:12 -- Started
2018-04-13 22:04:12 -- Started
2018-04-13 22:10:12 -- Started
2018-04-13 22:16:12 -- Started
2018-04-13 22:22:12 -- Started
2018-04-13 22:28:12 -- Started
2018-04-13 22:34:12 -- Started
2018-04-13 22:40:13 -- Started
2018-04-13 22:46:12 -- Started
2018-04-13 22:52:12 -- Started
2018-04-13 22:58:12 -- Started
2018-04-13 23:04:12 -- Started
2018-04-13 23:10:13 -- Started
2018-04-13 23:16:12 -- Started
2018-04-13 23:22:12 -- Started
2018-04-14 00:28:12 -- Started
2018-04-14 00:34:12 -- Started
2018-04-14 00:40:12 -- Started
2018-04-14 00:46:12 -- Started
2018-04-14 00:52:12 -- Started
2018-04-14 00:58:12 -- Started
```
You see there are 6 minutes between restarts (1 minute job + 5 minutes alarm delay)
This was run on battery only with a full Raspbian with desktop and VNC server active on a Pi3B
(not the most energy efficient configuration)

Notes:
- Only the pijuice module is needed for this. You could disable the pijuice.service and make sure you boot to command line with autologin (no Desktop/GUI).
- This scenario will only work while on battery or with power connected to the PiJuice.

closed 2018-04-24 by odeckmyn