PiSupply/PiJuice #92: pijuice-gui fails to install correctly and run

pettefar, 2018-04-12:
I am having a problem getting the pijuice-gui to run, the pijuice settings menu option does nothing and nothing is in the task bar.  If I run it manually it fails: 

root@beer:~# python /usr/bin/pijuice_gui.py 
Traceback (most recent call last):   File "/usr/bin/pijuice_gui.py", line 22, in      from Tkinter import Button as tkButton   File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 42, in      raise ImportError, str(msg) + ', please install the python-tk package' ImportError: libtk8.6.so: cannot open shared object file: No such file or directory, please install the python-tk package 
root@beer:~# apt-get install python-tk 
Reading package lists... Done Building dependency tree        Reading state information... Done python-tk is already the newest version (2.7.13-1). 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded. 
root@beer:~#   
root@beer:~# python3 /usr/bin/pijuice_gui.py 
Traceback (most recent call last):   File "/usr/lib/python3.5/tkinter/__init__.py", line 36, in      import _tkinter ImportError: libtk8.6.so: cannot open shared object file: No such file or directory  During handling of the above exception, another exception occurred:  Traceback (most recent call last):   File "/usr/bin/pijuice_gui.py", line 32, in      from tkinter import Button as tkButton   File "/usr/lib/python3.5/tkinter/__init__.py", line 38, in      raise ImportError(str(msg) + ', please install the python3-tk package') ImportError: libtk8.6.so: cannot open shared object file: No such file or directory, please install the python3-tk package 
root@beer:~# apt-get install python3-tk 
Reading package lists... Done Building dependency tree        Reading state information... Done python3-tk is already the newest version (3.5.3-1). 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded. root@beer:~#

pettefar, 2018-04-12:
Raspberry Pi 3.  Linux beer 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux

tvoverbeek, 2018-04-13:
The problem is the missing shared library libtk8.6.so
Something seems to be wrong with your tk installation.
The python-tk package has the libtk8.6 package as a dependency
libtk8.6 contains the shared library /usr/lib/arm-linux-gnueabihf/libtk8.6.so.
Can you check what you have in /usr/lib/arm-linux-gnueabihf/ ?
Maybe you have installed some other software which replaced libtk8.6 with a different version.

pettefar, 2018-04-13:
nick@tart ~ $ ls -al /usr/lib/arm-linux-gnueabihf/libtk*
-rw-r--r-- 1 root root 1073368 Nov  5  2014 /usr/lib/arm-linux-gnueabihf/
libtk8.5.so
lrwxrwxrwx 1 root root      11 Nov  5  2014
/usr/lib/arm-linux-gnueabihf/libtk8.5.so.0 -> libtk8.5.so
-rw-r--r-- 1 root root 1110376 Sep  2  2014 /usr/lib/arm-linux-gnueabihf/
libtk8.6.so
lrwxrwxrwx 1 root root      11 Sep  2  2014
/usr/lib/arm-linux-gnueabihf/libtk8.6.so.0 -> libtk8.6.so

Regards,

Nick Pettefar

pettefar, 2018-04-13:
Apologies, wrong Pi! Here is the one with the problem:

nick@beer:~ $ ls -al /usr/lib/arm-linux-gnueabihf/libtk*
-rw-r--r-- 1 root root 1143132 Aug  2  2016 /usr/lib/arm-linux-gnueabihf/
libtk8.6.so
lrwxrwxrwx 1 root root      11 Aug  2  2016
/usr/lib/arm-linux-gnueabihf/libtk8.6.so.0 -> libtk8.6.so

Regards,

Nick Pettefar

tvoverbeek, 2018-04-13:
You obviously have libtk8.6 in the right place.
How do you succeed to run under python3?
The pijuice-base package only installs for python2.7.
When I am running `python3 /usr/bin/pijuice_gui.py` I get:
```
pi@rpi-stretch-full:~ $ python3 /usr/bin/pijuice_gui.py
Traceback (most recent call last):
  File "/usr/bin/pijuice_gui.py", line 42, in <module>
    from pijuice import (PiJuice, pijuice_hard_functions, pijuice_sys_functions,
ImportError: No module named 'pijuice'
```

pettefar, 2018-04-13:
Hi Ton,

How does that address my problem?  Anyway, it has #! /usr/bin/python as the
first line so I guess it just runs the default python:
  nick@beer:~ $ python --version
  Python 2.7.13

Regards,

Nick Pettefar

tvoverbeek, 2018-04-13:
The only thing I can suggest is to remove python-tk and libtk8.6 and then install them again.
```
sudo apt-get purge libtk8.6
sudo apt-get purge python-tk
```
Check that libtk8.6.so is removed.
```
sudo apt-get install python-tk
```
This should also install libtk8.6.
If that does not help then I am out of suggestions.

tvoverbeek, 2018-04-13:
Glad it works now.
Closing issue now.

closed 2018-04-13 by tvoverbeek