PiSupply/PiJuice #93: Pogo pin too small?

NdK73, 2018-04-13:
Hello.
Seems the pogo pin is undercalibrated for RPi2B.
See photo.
Moreover, when I try to change to "Run pin: INSTALLED" from TUI and apply changes, it returns to "NOT_INSTALLED".
[image]

NdK73, 2018-04-13:
Here it is. Taken w/ USB microscope to highlight the details.
[image]

tvoverbeek, 2018-04-15:
Some more info. Compared my RPi2B, RPi3B and RPI3B+.
Indeed for the RPI2B the Pogo pin goes through the hole.
For the RPI3B and RPI3B+ the Pogo pin rests in the Run hole and does not go through.

shawaj, 2018-05-17:
@tvoverbeek thanks for the info.

@NdK73 what you will need to do is solder the pin to the PiJuice board, and then put a solder blob in the Pi2 hole. There isn't really another solution I am afraid.

shawaj, 2018-05-17:
@ChristopherRush shall we add this to the documentation?

I would like to test this with a few Pi2s just to make sure it is a issue for all of them

NdK73, 2018-05-17:
Instead of modifying the Pi, I think I'll just "silver" the pogo pin head with a veil of tin, so it "grows" that bit that won't let it fall through the hole...
Surely that a workaround is needed must be in the docs.

shawaj, 2018-05-17:
Yep @NdK73 good idea and yes we will add to docs

EricS61, 2018-05-17:
There's three parts to the pogo pin. The very thin part sits in the PiJuice (TP3), the middle part sits between the PiJuice and the Pi2, and the third piece which is adjustable with a loaded spring goes through the Pi2 like in the picture. BUT, if you look closely, the bottom of the middle piece touches (grounds) to the RUN on the Pi. I have two Pi2 Model B's that have the PiJuice w/pogo pin installed, and I'm having no problems. Pogo pin is showing as being installed when I bring up the PiJuice App in Raspbian Stretch.

ChristopherRush, 2018-05-17:
@shawaj Will update the docs to reflect what @NdK73 suggested.

shawaj, 2018-12-11:
@ChristopherRush did you manage to sort this?

ChristopherRush, 2018-12-11:
The pogo pin must have been on odd size. The latest PiJuice v2.2.2, the
pogo pin fits perfectly with the Pi 2B+ as tested.

NdK73, 2018-12-11:
I used a very thin wire. Just enough to have the pogo pin not fall throught. This way I didn't permanently modify anything.

shawaj, 2018-12-12:
@ChristopherRush do you mean 2B or 3B+? If it works with 2B that's cool.

I'm just wondering if we could just mention that if going through the hole we could just either tin the end of the Pogo pin or the hole itself. Just in case this issue is maybe to do with manufacturing tolerances or something?

ChristopherRush, 2019-09-15:
Documented for the next commit but pogo pin no longer required now.

closed 2019-09-15 by ChristopherRush