PiSupply/PiJuice #95: Test firmware V1.2 reports itself as V1.1

tvoverbeek, 2018-04-14:
See subject. Very confusing since I have a hard time to remember which firmware version I ahve loaded during various tests.
@mmilann please fix the version reporting asap

klint-k, 2018-04-14:
can we have a check for new firmware? we may just want to up dat the firmware.

tvoverbeek, 2018-04-15:
Normally firmware updates are distributed with an updated version of the pijuice raspbian packages.
After you update the package(s) pijuice_gui and pijuice_cli will tell you there is a new version available when you select the Firmware tab/menu.
This 1.2 version is not official yet and I would not recommend to use it yet.

ianjukes, 2018-04-23:
@shawaj I am having this same issue. I downloaded the 1.2 firmware from GitHub and have tried updating from (a) pijuice_cli.py and (b) pijuiceboot directly. The update seems to work using both methods, but then pijuice_cli.py reports firmware version is still 1.1 and that the update is still needed.

shawaj, 2018-05-17:
@tvoverbeek @ianjukes @klint-k please try the latest v1.2 firmware from https://github.com/PiSupply/PiJuice/tree/master/Firmware

This will also be available directly in the PiJuice software in the latest v1.3.3 update which is uploaded to the Raspbian repository right now but will take a few days to filter through into apt.

Will close this now as it is working for me. If you have any further issues, please post back here and we can re-open it if necessary.

closed 2018-05-17 by shawaj

tvoverbeek, 2018-05-17:
@shawaj Already tried it as soon as it was available on GitHub. Works for me.

shawaj, 2018-05-17:
Thanks Ton! Chris and Ilya also tried it and it works

ianjukes, 2018-05-29:
@shawaj Same here. Tried it as soon as it was up on GitHub and works like a charm. Thanks guys.