PiSupply/PiJuice #96: epoch time from pijuice ckock.

klint-k, 2018-04-14:
I know verry little about hwclock. 
Is there a way to get raspi epoch time from the pijuice directly? with out coversions.
 Klint

tvoverbeek, 2018-04-15:
Short answer: No
In Linux/Unix the system clock is always kept in UTC. When requesting to show the time (i.e. the date command in a terminal window, or the clock on the Desktop) the UTC time is converted to local time using the timezone information. That is why Daylight Saving/standard transitions are handled automatically (provided the timezone info is up to date).
Most PCs have a hardware clock to keep the time when the PC is off. At shutdown the system time is copied to the hardware clock and at boot the system clock is initialized from the hardware clock.
In Linux/Unix the hardware clock is kept in UTC.
The RPi does not have a hardware clock (or RTC) but several external ones are available.
Part of the PiJuice microcontroller is such a RTC and is kept in UTC.
The alarm functions (i.e. wakeup the RPi) are executed by the PiJuice microcontroller and hence have to be specified in UTC.

shawaj, 2018-05-18:
@tvoverbeek do you think this is something worth exploring or should we just close?