PiSupply/PiJuice #97: Questions about PiJuice functionning and configuration

sakura-lain, 2018-04-22:
Hi !

I am new to PiJuice (received mine yesterday from French supplier Kubii) and to GitHub, so I hope this is the good place to ask my questions, as I’m trying to configure PiJuice and think it is not so easy. I hace already read the documentation several times, and also the topics in Github’s discussion. I don’t think my questions were answered, but I am sorry if I missed something. Here are the problems I encounter :

*About EEPROM :

- First of all, I cannot see EEPROM at address 50 or 52 when I launch i2cdetect in a console or a terminal. Is it normal ? Is there another way to check whether EEPROM is present and properly working ?

- I am also unsure on how to activate/deactivate EEPROM. Is it via the unprotection of EEPROM write, or is it another function I’ve missed ?

*About buttons and functions :

- I cannot manage the buttons. Apart from the default functions HARD_FUNCTION_POWER_ON and HARD_FUNCTION_POWER_OFF on SW1, nothing else seem to work :

- I tried to program simple scripts (light the user led via a python script or launch xterm via a bash one) on SW2 and SW3 with SINGLE_PRESS (adjusted to 400 millis and 800 millis respectively) without any success. Tried to change permissions to 777, launch by adding the command python2 or source before the scripts, nothing seems to work. I could check the scripts launch properly via a console or a terminal. Also tried with reboot, shutdown or power off functions with no result.

- A LONG_PRESS1 with SW1 is by default supposed to launch SYS_FUNC_HALT. However, it doen’t seem to do anything. What is this function supposed to do ? Thought it should put the Pi in stand-by...

- pressing SW1 to shutdown or switch on the system only works when Pi’s on battery. Is it normal ? How can I make this button active when the Pi’s connected to electrical outlet ?

- For the user scripts is it compulsory to use a script or is it possible to directly program a bash command line as one of the USER_FUNC ?

- After having lit up the user led through a simple python script, I could remark that the led stays enlighted even after Pi’s reboot through console, as if rebooting the Pi didn’t make the script stop, whereas it is stopped if I shutdown the Pi through PW1 button. I suppose this is because the script execution is stored in the PiJuice’s microcontroller. Is there a way to make it stop at Pi’s shutdown through console or x-window system ?

*About powering :

- I don’t really get the difference between connecting the Pi to an electrical outlet through the Pi’USB or through the PiJuice’s one.

- The PiJuice, even disconnected from the Pi, keeps on consuming charge (its system led is blinking). Is there a way to really shutdown it, apart from removing the battery ? 

*About making working the Sensehat with PiJuice :

- Last but not least, did someone try to make a Sensehat work with PiJuice ? Although they seem compatible (peculiarly not using the same i2c adresses), could someone confirm they really are ? Is there something special to do to configure it ?

Moreover, I would like to point out what seems to be little bugs :

- when trying to light a led with several colors at 255 (i.e yellow, violet and white), only the red color lights up, with a strong intensity, as if the signal for the other colors were sent to that unique led. This problem may be linked to the one signaled by another contributor saying it was hard to distinguish between orange and red on first led.

- PiJuice Hat Configuration software is freezing my Pi’s x-window system when I apply changes. I have to reboot the whole system in this case through console. I am using the last 3B+ Pi with up-to-date Raspbian.

- I tried to consult tutorials at this address given on GitHub, but received the web page seems to be down :  https://learn.pi-supply.com/make/ 

Thank you for your work and for your answers and sorry again for the long list of questions.

tvoverbeek, 2018-04-22:
Here some answers:
EEPROM: This is the ID eeprom with configuration info for the HAT. It is on a different i2c bus (i2c-0)
hence you will not see it with the usual i2cdetect. i2cdetect should see the PiJuice at 0x14 and the RTC on the PiJuice at 0x68.
The HAT specification says the configuration EEPROM must be at address 0x50.
If you stack e.g. a sense hat on the PiJuice you have a conflict since both have an eeprom at 0x50.
If you want to use the sense hat you have to change the PiJuice eeprom address to 0x52 to avoid the conflict.
Removing the write protection of the ID eeprom is only needed if you want to reprogram it (not needed for normal users, leave it at write protected)

Regarding the buttons, they need the cooperation of the RPi itself. The pijuice.service has to be running otherwise only the HARD functions work (they are handled directly by the PiJuice).
Check with 'systemctl status pijuice.service' in a terminal.
Did you update the firmware to 1.1? There are problems with the 1.0 firmware (the delivered version).

The pogo pin is only needed when powering via the RPi. The PiJuice hardware was designed before the 3B+ was available. Now there are already 3 different locations for the pogo pin:
TP1 - Pi 3B
TP2 - Pi Zero
TP3 - Pi B+ and 2B

When powering via the PiJuice the power to the RPi can be completely removed.
When powering via the RPi the RPi still draws power even when shut down (think off external USB units powered via the 5V USB)

For your SSH problem I assume you lose the SSH connection to the Pi since there is no way you can SSH to the PiJuice. Most likely a power supply problem. Note the 3B+ needs more power than previous models, you really need a good 2.5A power supply.

sakura-lain, 2018-04-22:
Thank you for your answer ! I'll try that and come back to share the results.

sakura-lain, 2018-04-22:
Hi again !
So I got my Sensehat working :smiley: , but I'm still stuck on the button part :disappointed:. Checked out 'systemctl status pijuice.service', it's active. Anything else I could try ?

Thanks again for your help.

tvoverbeek, 2018-04-22:
What is the version of your pijuice packages?
Look at the output of 'apt-cache show pijuice-base'.
The latest version in Raspbian is 1.3.2-1
If your version is older, please update ('sudo apt-get update, sudo apt-get upgrade').

sakura-lain, 2018-04-22:
Hi again ! This is the right version (I've upgraded yesterday). 

Moreover, the Sensehat seems unstable. After a shutdown it didn't work anymore. I tried to reboot, unpluged/repluged the Sensehat but finally got to cut off battery connection, restart and after that it works again. When pluging/unpluging the Sensehat, I got error messages linked to EXT4fs. Maybe should I try reinstalling Raspbian?

sakura-lain, 2018-04-22:
Hi ! So I fully reinstalled Raspbian and now the software works like a charm, no more freezes. I finally could manage the buttons : I simply forgot to put a delay to recognize they are pressed. I have three more questions anyway:

- I have to reboot the system so that the modifications are taken into account after applying them through the software. Is there a way to reset them in a simpler way?

- I remarked a delay of nearly one second between the time a button is pressed or released and the time it is taken into account by the PiJuice. Is it normal ?

- I kept the default functions on SW1 (just lowering the delays for LONG_PRESS 1 and 2) but still don't get the difference what SYS_FUNC_HALt is supposed to do, I don't feel it's working...

- Using the Sensehat with the PiJuice still works randomly, despite having modified the EEPROM i2c address.

Anyway, it's mostly working! Thank you for your help!

tvoverbeek, 2018-04-23:
Taking your bullets in order:
```
- I have to reboot the system so that the modifications are taken into account after applying them
  through the software. Is there a way to reset them in a simpler way?
```
This is a known problem, which is already fixed on github. I cannot tell you when the next update to the Raspbian packages will be published. @shawaj or @mmilann might be able to tell you.
```
- I remarked a delay of nearly one second between the time a button is pressed or released and the
  time it is taken into account by the PiJuice. Is it normal ?
```
The pijuice.service running on the RPi polls the PiJuice board once a second for info like button presses, low voltage, etc. Hence the reaction time of the software can be up to 1 second.
```
- Pressing a button to launch a function changing the user led color doesn't seem to work if another
  function is launched on that led. For instance, I have functions on SW3 that light the led in blue or
  green when the button is pressed or released, and a function on SW2 that makes the led blink when
  pressed. But pressing SW3 doesn't stop the blinking. Is there a way to do that?
```
When you send a LED command, the RPi sends the command to the PiJuice microcontroller. This
executes the command and will not accept an other LED command until the previous one is finished.
In your case you must have specified a large (or infinite) number of blink cycles.
From the github software page README:
______
**SetLedBlink(led, count, rgb1, period1, rgb2, period2)**
Plays blink pattern on LED configured as “User LED”. Arguments: led: LED designator, one of: 'D1', 'D2'. count: number of blinks for count in range [1 - 254], blink indefinite number of times for count = 255. rgb1: [r, g, b] is array of brightness levels of LED components in first period of blink, where r, g and b, are in range [0 – 255]. period1: duration of first blink period in range [10 – 2550] miliseconds. rgb2: [r, g, b] is array of brightness levels of LED components in second period of blink, where r, g and b, are in range [0 – 255]. Period2: duration of second blink period in range [10 – 2550] miliseconds. 
_____
```
- I kept the default functions on SW1 (just lowering the delays for LONG_PRESS 1 and 2) but still don't 
  get the difference what SYS_FUNC_HALT is supposed to do, I don't feel it's working...
``` 
SYS_FUNC_HALT when executed after LONG_PRESS_1 blinks LED2 and invokes 'sudo halt' on the Pi which you can recognize with the 10 flashes of the green RPi LED at the end of the shutdown. Power to the RPi is not removed: RPi red power LED stays on. When power is applied via the PiJuice USB a short press of SW1 should then reboot the Pi (Pi Power is shortly interrupted to start the boot sequence). When power is applied to the Rpi you need the pogo pin to be able to boot.
Other interesting function is SYS_FUNC_HALT_POW_OFF: Also shuts down the Pi and removes the power to the Pi (Red power LED on the Pi goes off) after 60 seconds.
```
- Using the Sensehat with the PiJuice still works randomly, despite having modified the EEPROM i2c
  address.
```
One thing to consider is the distance between the PiJuice and the bottom components of the Sense
Hat. I would personally use an [extension header](https://shop.pimoroni.com/products/booster-header) to increase the distance between the PiJuice and the Sense Hat.
```
- When on battery, my screen sometimes blink, I think it's the same problem as the one with the SSH
 I mentioned in my first post. Probably a powering problem, as you suggest, but maybe a bug too.
```
Cannot really help, but might be a power issue since you need power for both the 3B+ and the Sense Hat. When using the BP7X battery only, max current which can be supplied is about 1.1A. Might be marginal to power both the 3B+ and the Sense Hat.

For your last 2 points I have nothing to add, except for the quick discharge, see my power remarks above.

rossirm, 2018-07-10:
Hi to all!
I am Rossi from Bulgaria and began to use PiJuice 2 weeks ago.
I have a system of Raspberry Pi 3B model + PiJuice + Sense Hat. All works well instead of the fact that I can't say to button SW3 for example to execute a python script which is in the directory **/home/pi**
I show the path to the PiJuice software/User Scripts like /home/pi/python3 file_name.py
But the result is that when I press button SW3 it doen's work although I used commands in the terminal chmod +x file_name.py and sudo 777 /home/pi/file_name.py
PLEASE HELP ME! I want it to work very much.
Perhaps this is not so big problem for you, but I am new to Raspberry world and sorry for my English if it is not so good!

shawaj, 2018-07-10:
@rossirm we will reply on #168 instead of here

rossirm, 2018-07-10:
Thanks I will wait for your help. 
What do I have to write at the beginning of my script file. Is it `#!/usr/bin/python3` or something else?

closed 2019-09-15 by ChristopherRush