PiSupply/PiJuice #100: Strange battery charging behavior

sakura-lain, 2018-04-23:
Hi again,

I'm keepin on testing the PiJuice and remark a strange behavior using the pijuicetest.py provided here : https://github.com/PiSupply/PiJuice/blob/master/Software/Test/pijuicetest.py

My battery seems to charge normally until around 70%, no more. Beyond, it stays stuck at 71-72% (oscillating between the two values). I'm testing it : after a short unplug-replug, it fell to 67% then right now after that 73%, then stuck at 76%. So I unplugged and had a 77%, replugged and it just fell back to... 14% (with the led turning red for a few ten of seconds) ! The led green again, it oscillated between 78 and 80%. It's now most of the time 82%, after a phase around 84-85%. It has just displayed a jump at 95% (I'm spamming the script several times a try) before falling to 82% again. So I unplugged to check, it showed 63%... Replugged : 57%, then 94%, 78%, then 84% again; then back to 79%, and now 96% (the whole thing unplug-replug and displaying those results took me maybe three minutes)!

This is quite strange, but I've no idea whether it's a hardware problem or a software problem. Is the battery correctly charging but the indicators are false? Is it  discharging and charging at the same time, so its charge level cannot move or can hardly increase (it may be curious, since until 70% it used to charge very quickly and regularly)? Are there peaks of consumption (especially when unplugged) that could partially explain those variations? Can't my battery fully charge (this is the one provided with the PiJuice, so it si new)? Any ideas? Did someone remark something similar? I also noticed that when the PiJuice is plugged, the blue blinking for the charge never stops : is it normal or is it the sign that it doesn't charge correctly? 

I'm ending here at 96% (soon after displays at 97%), so my favorite hypothesis is that it's charging but for some reason indicators are false. I'll check again later anyway to see how it evolves (oh no: 81%!).

Thank you for your help!

sakura-lain, 2018-04-23:
Time to post, it's 99%! Finally doing it, but the reason of this behavior remains unexplained...

sakura-lain, 2018-04-23:
So after a reboot it shows 100% (sometimes but rarely 99%) on both console and battery indicator on graphical interface. Anyway, the charging led is still blinking blue. Both show 99% when unplugged.

sakura-lain, 2018-04-23:
Oh, it fell down to 75% in a few seconds, whereas the icon indicating the battery level is still showing full. Seems now discharging almost normally even if quite quickly and a bit irregularly (69%-72%-65%-68% in less than one minute, battery icon still full).

sakura-lain, 2018-04-23:
Oscillating between 61 and 63%, the discharging rhythm seems more normal now. Still quick, but not shocking...

sakura-lain, 2018-04-23:
54%, led green again... There's surely is a problem somewhere, and seems I'm not the only one to encounter this.

sakura-lain, 2018-04-23:
44%, etc.

sakura-lain, 2018-04-23:
26%, led red, battery icon full... :open_mouth:

ChristopherRush, 2018-04-23:
Charge level measurement precision is decreased by increased load current due to internal resistance of protection circuit inside BP7X battery. It is like logarithmic dependence, so there can be more than 50% charge when it is actually around 10% if load current is enough high.

sakura-lain, 2018-04-23:
So is there no way to really know the actual level of charge ? After an hour of discharging, I'm now at 6%, icon still showing a full battery. What should I conclude? Is there a way to calculate the real battery level?

sakura-lain, 2018-04-23:
So it went for shutdown around 20h, soon after I launched my Sensehat weather station. Think with a Pi 3B+, it can stayed on for maybe 3 to 4 hours when doing nothing special, but discharging soon when having the Sensehat active (think I measured around 1 hour at full charge, to be confirmed has I didn't make a precise measure). This confirms also that in my case, battery level indicators are not accurate at all.

ChristopherRush, 2018-04-26:
What was your average load mAh? An hour seems fair at around 1A-1.4A load and 4 hours idle is what it should be. As mentioned before the battery levels are based on current consumption at that time, giving real-time battery levels.

QL8-6754, 2018-04-27:
Fully charged the battery and started my Pi3B with a clean 2018-04-18 Raspbian Desktop microSD card. After a long period of charging the PiJuice software indicated 100%. Upon removing power the indicator went to 99% (which is fine). With the Pi under no additional load (ie nothing apart from the OS was running - Bluetooth had been switched off - wifi was on) the battery level dropped to 80% within about 20 seconds and steadily ticked down. When the level crossed under 50% the led went red rather than amber. Within an hour the power level was reading 4%. The taskbar indicator showed a fully charged device however hovering the mouse over the icon showed the power level shown in the main settings program. I intended to experiment further but it is possible that a Raspbian update borks the PiJuice software, making monitoring the power impossible ([see here](https://github.com/PiSupply/PiJuice/issues/104)).

woodentop, 2018-04-30:
See #108 for mismatch between icon and battery level.

shawaj, 2018-12-08:
Check out the beta v1.3 firmware https://github.com/PiSupply/PiJuice/tree/master/Firmware/Beta

It shouldn't brick your PiJuice, but it might cause some odd behaviour so it is not recommended for use in production environments and you do so at your own risk.

More here - https://github.com/PiSupply/PiJuice/issues/249

ChristopherRush, 2019-09-15:
Fixed in firmware v1.3 - f571b4384aea02c75073204db5213538a9c02fbf

closed 2019-09-15 by ChristopherRush